wpilibsuite/shuffleboard #41: Option to not show full NetworkTable path

AustinShalit, 2017-07-19:
There should be an option to only show the last part of a networktable path

[image]

nightpool, 2017-07-19:
This used to be the behavior. I'm not exactly sure why it changed, but
yes—this is too much information by default.

SamCarlberg, 2017-07-19:
@nightpool It never behaved like that. We could do this, but I'm not sure how useful it would be. There are almost never empty subtables like this unless ntcore is doing the initial server sync

bradamiller, 2017-07-19:
I had made the same comment a few weeks ago. Maybe hovering over it could show the full path.

bradamiller, 2017-07-19:
@SamCarlberg, I'm confused about your comment above. The "SmartDashboard" part of the path almost always obscures single tile widgets.

SamCarlberg, 2017-07-19:
I was talking about the tree

nightpool, 2017-09-06:
We can just show the leaf key name when after https://github.com/wpilibsuite/shuffleboard/issues/171 is implemented.

SamCarlberg, 2017-11-20:
Fixed by #198

closed 2017-11-20 by SamCarlberg