wpilibsuite/shuffleboard #42: Need a minimal UI for the DriverStation

AustinShalit, 2017-07-19:
You should be able to hide everything except the widget tab for use on a driver station

SamCarlberg, 2017-07-19:
So basically hide the sources pane? That's on the roadmap

bradamiller, 2017-07-19:
Ideally, resize the dashboard to fill the remaining space on the screen when in that mode. So the driver station starts first, takes it's bit of screen real estate, then when in competition mode (or whatever), this comes up and fills the rest of the screen. This works since the DS is fixed size.

andrewda, 2018-02-22:
I'd love to use Shuffleboard for the Driver Station, but I feel like this needs to be done first. When can we expect a driver mode to hide the sources and widgets and only display the current layout?

SamCarlberg, 2018-02-22:
You mean have shuffleboard automatically do that, instead of manually moving the slider over?

andrewda, 2018-02-22:
Yea, having it be an option which can be remembered between restarts would be super nice.

SamCarlberg, 2018-02-22:
It's already saved as part of the layout. If you move the slider over and save, that position will persist

andrewda, 2018-02-22:
Sweet okay, my apoligies then I didn't know that was persistent!

UserC2, 2024-02-22:
> Sweet okay, my apologies then I didn't know that was persistent!

This issue should likely be closed, as it seems it was resolved.