wpilibsuite/shuffleboard #44: Do not use the SmartDashboard directory

AustinShalit, 2017-07-19:
People may want to still use SmartDashboard.  Are we sure what we put in the directory will not conflict?

SamCarlberg, 2017-07-19:
The plan is to replace the old dashboard with this one and stop supporting it. But there shouldn't be any conflicts, the only stuff that gets put in that directory by the old dashboard are the `save.xml` files and extension JARs. Shuffleboard is putting recording files and `save.json`

bradamiller, 2017-07-19:
This seems somewhat presumptuous. Maybe we should wait until this is up, running, and people are happy with it before totally abandoning the old dashboard.

nightpool, 2017-07-19:
I agree with brad, but either way I don't see any possible conflicts or
problems with reusing the old directory.

bradamiller, 2017-07-19:
as long as it's using the same directly in a non-conflicting way.

SamCarlberg, 2017-07-19:
There are no filename conflicts

PR #306 (2017-12-21): General housekeeping cleanup
Fixes lots of minor issues:

- Fixes #44
- Fixes #115
- Fixes #127
- Fixes #142
- Fixes #221
- Fixes #264
- Fixes #272

Also improves performance of previewing tiles from dropped sources by caching the size rather than creating and computing the size of a new widget/layout EVERY time the mouse is moved while dragging
commit ead9bfb8 (2017-12-21): General housekeeping cleanup (#306)

* Rename widget classes to end with "Widget", rename WidgetPropertySheet

Fixes #142
Fixes #264
Fixes #272

* Use ~/Shuffleboard instead of ~/Smartdashboard

Replaces most instances of "Smartdashboard" to "Shuffleboard", except for the Smartdashboard tab since that contains stuff in the Smartdashboard table from the robot
Fixes #44

* Show grid highlight when resizing tiles

Fixes #127

* Preview size of dropped widgets from the gallery

Also improves performance of previewing tiles from dropped sources by caching the size of the widget rather than creating [... cut]
closed 2017-12-21 by bradamiller