wpilibsuite/shuffleboard #49: Launches sized too large

Kevin-OConnor, 2017-07-19:
Running shuffleboard on my machine results in it launching with the window size too large. It flows off the top and bottom of the screen, hiding the menus.

SamCarlberg, 2017-07-19:
Operating system?

Kevin-OConnor, 2017-07-19:
Windows 7 x64. I also just noticed a few errors in the Gradle window, not sure if they are related or not:
Exception in thread "JavaFX-Launcher" java.nio.file.InvalidPathException: Illegal char <:> at index 66: C:\Users\koconnor/SmartDashboard/dashboard_recording_2017-07-19_13:38:25.frc
        at sun.nio.fs.WindowsPathParser.normalize(WindowsPathParser.java:182)
        at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:153)
        at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:77)
        at sun.nio.fs.WindowsPath.parse(WindowsPath.java:94)
        at sun.nio.fs.WindowsFileSystem.getPath(WindowsFileSystem.java:255)
        at java.nio.file.Paths.get(Paths.java:84)
        at edu.wpi.first.shuffleboard.sources.recording.Serialization.saveRecording(Serialization.java:107)
        at edu.wpi.first.shuffleboard.sources.recording.Recorder.saveToDisk(Recorder.java:60)
        at edu.wpi.first.shuffleboard.sources.recording.Recorder.lambda$new$98(Recorder.java:41)
        at edu.wpi.first.shuffleboard.sources.recording.Recorder$$Lambda$82/18330560.changed(Unknown Source)
        at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(ExpressionHelper.java:182)
        at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
        at javafx.beans.property.BooleanPropertyBase.fireValueChangedEvent(BooleanPropertyBase.java:103)
        at javafx.beans.property.BooleanPropertyBase.markInvalid(BooleanPropertyBase.java:110)
        at javafx.beans.property.BooleanPropertyBase.set(BooleanPropertyBase.java:144)
        at edu.wpi.first.shuffleboard.sources.recording.Recorder.setRunning(Recorder.java:146)
        at edu.wpi.first.shuffleboard.sources.recording.Recorder.start(Recorder.java:94)
        at edu.wpi.first.shuffleboard.Shuffleboard.init(Shuffleboard.java:35)
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:841)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$152(LauncherImpl.java:182)
        at com.sun.javafx.application.LauncherImpl$$Lambda$50/15491924.run(Unknown Source)
        at java.lang.Thread.run(Thread.java:745)
Jul 19, 2017 1:38:26 PM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 8.0.111 by JavaFX runtime of version 8.0.45
Jul 19, 2017 1:38:29 PM com.sun.javafx.css.StyleConverterImpl getInstance
SEVERE: StyleConverterImpl : converter Class is null for : com.sun.javafx.scene.layout.region.CornerRadiiConverter
Jul 19, 2017 1:38:29 PM com.sun.javafx.css.StyleConverterImpl readBinary
SEVERE: could not deserialize com.sun.javafx.scene.layout.region.CornerRadiiConverter could not deserialize com.sun.javafx.scene.layout.region.CornerRadiiConverter

Kevin-OConnor, 2017-07-19:
Display #1 is 1920x1080. Display #2 is to the left of it (extended desktop) and is 1920x1200.

SamCarlberg, 2017-07-19:
Weird. @calcmogul had the same problem on Linux with Xfce at 1366x768, but it works for me with Pantheon at 1920x1080, and it seems to work on OSX.

The stacktrace is from playback, looks like Windows doesn't allow colons in file names...

SamCarlberg, 2017-08-14:
@Kevin-OConnor Can you test this again on `master`?

Kevin-OConnor, 2017-08-14:
Working properly for me on master.

closed 2017-08-14 by Kevin-OConnor