wpilibsuite/shuffleboard #101: Should publish standalone API jar

SamCarlberg, 2017-08-06:
A tentative list of the classes that should be moved into an `api` project:

| Package | Class |
|---|---|
| `components` | `LinearIndicator` 
|| `NumberField` |
||`Scrubber`|
| `data` | Everything in this package |
| `data.types` | Everything in this package |
| `properties` | `AsyncProperty` |
| `sources` | `DataSource` |
| | `AbstractDataSource` |
| | `SourceType` (should also be split into an interface and util class) |
| | `IncompatibleSourceException` |
| `widget` | `Widget`
| | `AbstractWidget` |
| | `AnnotatedWidget` |
| | `SingleTypeWidget` |
| | `ComplexAnnotatedWidget` |
| | `@Description` |
| | `@ParamtrizedController`

Other thoughts:

- Should this be a fatjar or standalone? A fat jar would give users access to guava etc. but would increase the jar size 
- Need to look into how plugins would interact with an API jar and determine any extra classes that should get moved over. Particularly custom JavaFX components.

@JLLeitschuh can you work on splitting the project into an `api` project and an `app` project?

JLLeitschuh, 2017-08-06:
I'd be happy to take this on.

Do you have an idea what the fully qualified destination package is you'd like these each to end up in?
`edu.wpi.shuffleboard.api`?

Also, are there any big PR's coming in that will be messing with any of these files? I don't really want to/have the time to resolve massive merge conflicts.

SamCarlberg, 2017-08-06:
This should be pretty smooth. Adding plugins is going to be a mess.

I was thinking `edu.wpi.first.shuffleboard.api` for the base package name.

PR #104 (2017-08-07): Splits app into API and APP project
Closes #101

This splits the project into two separate projects. One called `api` and the other called `app`. In the future the `api` will be published as an API that teams can write plugins against.
commit 2be82ad7 (2017-08-07): Splits app into API and APP project (#104)

* Splits app into API and APP project

Closes #101

* Split SourceType into class and util class (like DataType)
closed 2017-08-07 by SamCarlberg