wpilibsuite/shuffleboard #106: Widget UI could use some work (borders perhaps?)

JLLeitschuh, 2017-08-07:
The widgets tab kind of looks sad. It might help to have borders around each component.

[image]

As it is it the title isn't visually tied to the component.

Sorry if this is a duplicate issue.

SamCarlberg, 2017-08-07:
This is how it looks for me (on `master`)

[image]

JLLeitschuh, 2017-08-07:
The above image is what it looked like for me on master. I can try cleaning and rebuilding.

SamCarlberg, 2017-08-07:
Looks like the CSS stylesheets didn't load for you

JLLeitschuh, 2017-08-07:
I ran `./gradlew clean` and then `./gradlew run` (after fixing the `theMainClassName` like you did in your PR). It still looks like my picture above.

SamCarlberg, 2017-08-07:
The console should have errors saying the stylesheets can't be loaded. Do you see that?

nightpool, 2017-08-07:
@JLLeitschuh Make sure you're on c4d6565c, there were more changes related to style sheets in the PR. it sounds from your comment like you're on an older commit and just made the change manually.

AustinShalit, 2017-08-07:
master has borders for me

JLLeitschuh, 2017-08-08:
I'll check again tomorrow

JLLeitschuh, 2017-08-08:
Looks fine now.... Weird.

closed 2017-08-08 by JLLeitschuh

SamCarlberg, 2017-08-08:
Evan was correct: https://github.com/wpilibsuite/shuffleboard/issues/106#issuecomment-320783950