wpilibsuite/shuffleboard #124: Tab Preferences dialog starts too small

AustinShalit, 2017-08-10:
It starts with scroll bars now.  It should not.

AustinShalit, 2017-08-10:
Image for context: 
[image]

nightpool, 2017-09-06:
set fitToHeight=true from CSS

nightpool, 2017-09-17:
Oh, jesus, this was a hairy bug. Adding fitToHeight didn't do anything. 

To make a long story short, it turns out that PropertySheet doesn't handle setting padding correctly, at *all*. If you set any padding on PropertySheet, it ignores that completely when doing layout calculations.

But what was setting the padding in the first place? We never set padding on PropertySheet, and neither does ControlsFX (or they would have found this bug right away). Well, it turns out that, by default, Dialog panes set padding on their content. So by overriding that style, I got rid of the bug. It looks kinda terrible though, so I might just end up wrapping the PropertySheet in a BorderPane or something.

commit 0d2922bb (2017-09-18): Fix #124 (#185)

* Fix #124

When PropertySheet has padding, it doesn't calculate it's height correctly.
Dialogs give some padding to their content
So PropertySheets directly embedded in Dialogs are too short
closed 2017-09-18 by SamCarlberg