wpilibsuite/shuffleboard #126: :app:junitPlatformTestException in thread "JavaFX Application Thread" java.lang.IndexOutOfBoundsException

AustinShalit, 2017-08-10:
https://travis-ci.org/wpilibsuite/shuffleboard/jobs/263173372

@JLLeitschuh

JLLeitschuh, 2017-08-10:
If you look at the exception this is actually caused by `FxUtils` not Junit5.

This might be a concurrent modification exception in disguise? I don't really know, you guys are doing some crazy thread interweaving with the network tables listeners.

<details>
<summary>Really long exception below</summary>
<pre>
java.lang.IndexOutOfBoundsException
	at com.sun.javafx.scene.control.ReadOnlyUnbackedObservableList.subList(ReadOnlyUnbackedObservableList.java:136)
	at javafx.collections.ListChangeListener$Change.getAddedSubList(ListChangeListener.java:242)
	at javafx.scene.control.TreeTableView$TreeTableViewArrayListSelectionModel.handleSelectedCellsListChangeEvent(TreeTableView.java:3244)
	at javafx.scene.control.TreeTableView$TreeTableViewArrayListSelectionModel.access$2000(TreeTableView.java:2289)
	at javafx.scene.control.TreeTableView.sort(TreeTableView.java:1817)
	at edu.wpi.first.shuffleboard.app.widget.NetworkTableTreeWidget.makeBranches(NetworkTableTreeWidget.java:152)
	at edu.wpi.first.shuffleboard.app.widget.NetworkTableTreeWidget.lambda$null$7(NetworkTableTreeWidget.java:77)
	at java.util.HashMap.forEach(HashMap.java:1288)
	at edu.wpi.first.shuffleboard.app.widget.NetworkTableTreeWidget.lambda$new$8(NetworkTableTreeWidget.java:77)
	at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(ExpressionHelper.java:182)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.binding.ObjectBinding.invalidate(ObjectBinding.java:172)
	at org.fxmisc.easybind.monadic.FlatMapBindingBase.mappedInvalidated(FlatMap.java:76)
	at org.fxmisc.easybind.monadic.FlatMapBindingBase.lambda$new$22(FlatMap.java:21)
	at javafx.beans.WeakInvalidationListener.invalidated(WeakInvalidationListener.java:83)
	at com.sun.javafx.binding.ExpressionHelper$Generic.fireValueChangedEvent(ExpressionHelper.java:349)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.ObjectPropertyBase.fireValueChangedEvent(ObjectPropertyBase.java:105)
	at javafx.beans.property.ObjectPropertyBase.markInvalid(ObjectPropertyBase.java:112)
	at javafx.beans.property.ObjectPropertyBase.set(ObjectPropertyBase.java:146)
	at edu.wpi.first.shuffleboard.api.properties.AsyncProperty.lambda$set$2(AsyncProperty.java:54)
	at edu.wpi.first.shuffleboard.api.util.FxUtils.runOnFxThread(FxUtils.java:45)
	at edu.wpi.first.shuffleboard.api.util.AsyncUtils.runAsync(AsyncUtils.java:24)
	at edu.wpi.first.shuffleboard.api.properties.AsyncProperty.set(AsyncProperty.java:54)
	at javafx.beans.property.ObjectProperty.setValue(ObjectProperty.java:69)
	at edu.wpi.first.shuffleboard.api.sources.DataSource.setData(DataSource.java:60)
	at edu.wpi.first.shuffleboard.app.sources.CompositeNetworkTableSource.lambda$new$0(CompositeNetworkTableSource.java:49)
	at edu.wpi.first.shuffleboard.app.sources.NetworkTableSource.lambda$null$0(NetworkTableSource.java:50)
	at edu.wpi.first.shuffleboard.api.util.FxUtils.lambda$runOnFxThread$0(FxUtils.java:49)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.monocle.RunnableProcessor.runLoop(RunnableProcessor.java:92)
	at com.sun.glass.ui.monocle.RunnableProcessor.run(RunnableProcessor.java:51)
	at java.lang.Thread.run(Thread.java:748)
Exception in thread "main" java.lang.IllegalStateException: Not on FX application thread; currentThread = main
	at com.sun.javafx.tk.Toolkit.checkFxUserThread(Toolkit.java:236)
	at com.sun.javafx.tk.quantum.QuantumToolkit.checkFxUserThread(QuantumToolkit.java:423)
	at javafx.scene.Parent$2.onProposedChange(Parent.java:367)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:113)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:108)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.updateChildren(LabeledSkinBase.java:575)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.handleControlPropertyChanged(LabeledSkinBase.java:204)
	at com.sun.javafx.scene.control.skin.ListCellSkin.handleControlPropertyChanged(ListCellSkin.java:49)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase.lambda$registerChangeListener$61(BehaviorSkinBase.java:197)
	at com.sun.javafx.scene.control.MultiplePropertyChangeListenerHandler$1.changed(MultiplePropertyChangeListenerHandler.java:55)
	at javafx.beans.value.WeakChangeListener.changed(WeakChangeListener.java:89)
	at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(ExpressionHelper.java:182)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.StringPropertyBase.fireValueChangedEvent(StringPropertyBase.java:103)
	at javafx.beans.property.StringPropertyBase.markInvalid(StringPropertyBase.java:110)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:144)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:49)
	at javafx.beans.property.StringProperty.setValue(StringProperty.java:65)
	at javafx.scene.control.Labeled.setText(Labeled.java:145)
	at com.sun.javafx.scene.control.skin.ComboBoxListViewSkin.updateDisplayText(ComboBoxListViewSkin.java:375)
	at com.sun.javafx.scene.control.skin.ComboBoxListViewSkin.access$100(ComboBoxListViewSkin.java:57)
[... cut ...]
	at edu.wpi.first.shuffleboard.api.util.FxUtils.lambda$runOnFxThread$0(FxUtils.java:49)
	at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
	at com.sun.glass.ui.monocle.RunnableProcessor.runLoop(RunnableProcessor.java:92)
	at com.sun.glass.ui.monocle.RunnableProcessor.run(RunnableProcessor.java:51)
	at java.lang.Thread.run(Thread.java:748)
</pre>
</details>

nightpool, 2017-08-10:
yes this looked like a concurrentmodificationexception to me.

i was surprised that the test passed.....

JLLeitschuh, 2017-08-10:
Because the thread that threw the exception was not the thread that did the calling to schedule the runnable.

closed 2017-09-06 by nightpool