wpilibsuite/shuffleboard #138: Support for Undo/Redo moving widgets/ect

JLLeitschuh, 2017-08-11:
Designing for Undo/Redo generally means designing your software to support this from the beginning as you need to capture the actions along with a corresponding  function that can be called to undo the action.

Is there any plan to support this? If so, how would this work with plugins?

SamCarlberg, 2017-08-11:
Listening to JavaFX property changes will get us a long way. 

```java

class UndoRedo {

  public static void <T> recordChange(ObservableValue<T> property, T oldValue, T newValue) {
    ...
  }

  public static void record(ObservableValue<?> property) {
    property.addListener(undoRedo::recordChange);
  }

}

...

UndoRedo.record(someProperty);
```

Storing the changes in a `WeakHashMap<ObservableValue, Action>` and keeping a record of the order of actions could do the rest.

bradamiller, 2017-08-11:
Undo/Redo is usually done by breaking the actions that the UI can do into discrete actions. Then make an interface for a class that implements an action and create actions for each operation. Then add a do and undo method to the class. The action instances are added to the undo list as they are executed. If some actions can’t be undone, then clear the undo list before adding one of those.

JLLeitschuh, 2017-08-11:
I think you want to go with brads solution here.

Is this a feature that you want @bradamiller? Is it a priority?

I don't think that GRIP will ever support undo/redo because we didn't design it to support it from the beginning.

If this is something you want to support then you probably need to start designing for it now.

bradamiller, 2017-08-11:
I thought it would probably to much to do at this point unless you say it’s really easy. So the comment was mostly just a suggestion on how to do it.