wpilibsuite/shuffleboard #161: Changing team number/host doesn't clear NetworkTables

SamCarlberg, 2017-08-24:
When changing the NetworkTable server, entries are not cleared locally so every entry that existed on the previous host will be pushed to the new one

AustinShalit, 2017-08-24:
@PeterJohnson What do you think?

nightpool, 2017-08-25:
@samcarlburg this is probably the wrong issue for this project. If this is
the wrong behavior (which there appears to be some contention on) its a
fault of networktables, not us.

SamCarlberg, 2017-08-25:
It's on us to make sure we clean up properly

PeterJohnson, 2017-08-25:
The NetworkTable synchronization change will automatically clear anything that's not been written locally on connection to the server (if the server doesn't have the same key).  However, of it HAS been written locally, it will not be removed.  There are two questions here: 1. does the app write to values only when directed to by the user (which will make NT synchronization DTRT) and 2. do we want the app to clear everything out when the team number changes (not something NT keeps track of).  I think we want the answer to both of these to be yes.

PeterJohnson, 2017-08-25:
Note this has to be done fairly carefully to avoid corrupting the other end.  You should stop the client, wait for it to disconnect, clear everything (including persistent values), then start the client against the new ip.

PeterJohnson, 2017-08-25:
With the new instance API, this can be easier and safer depending on how your code is structured, as you can just create a new instance for the new team connection.

AustinShalit, 2017-08-25:
I think a good plan of action for solving this issue it to wait until the instance API is merged and then solve it by spinning up a new instance.

nightpool, 2017-08-25:
I would be wary about clearing the entire dashboard without prompting the user to at least save their layout. unless there's been a regression recently the answer to 1 is definitely yes.

I would generally expect most people to not have to change their host or team number, FWIW. Not sure how big of an issue this is.

SamCarlberg, 2017-08-25:
The dashboard isn't cleared when sources are removed. The widgets are disabled, but that's all.

nightpool, 2017-09-06:
This is currently a bug, and right now it has the workaround of restarting shuffleboard and connecting to the new host. 

If we start saving app preferences, this may become a bigger issue.

bradamiller, 2017-09-06:
Seems like we can let this go until after beta. We just need to warn teams that they have shut down shuffleboard before changing the team number. When it comes up, set the new team number.

Once we add the preferences saving, then it will come back up to the same host and the NT cache problem will exist.

SamCarlberg, 2018-01-15:
Fixed by #336

closed 2018-01-15 by SamCarlberg