wpilibsuite/shuffleboard #162: better default host

nightpool, 2017-08-25:
nearly noone will ever want to connect to localhost, we shouldn't optimize for it.

is there an mdns way to detect a robot automatically? could we use that value instead? otherwise, we should prompt for team number/host on startup if the value is empty

AustinShalit, 2017-08-25:
Right now I think it will connect to the same robot as whatever the driver station is connected to.  I will test tomorrow when I have access to a roboRIO.

ThadHouse, 2017-08-25:
NetworkTables itself will attempt to contact the robot directly by grabbing an IP from the currently running DS, although I don't know if this will happen directly now with the new instance stuff. It might have to explicitly be enabled, but I'd have to check.

closed 2017-09-06 by nightpool