wpilibsuite/shuffleboard #193: API jar needs to be published

SamCarlberg, 2017-09-19:
Right now, we're only publishing the app uberjar as `edu.wpi.first.shuffleboard:Shuffleboard:<version>`

We should publish two jars: an API jar (may be an uberjar to give users access to guava etc.) and the fat app jar.

| Project | Artifact name |
|---|---|
| API | `shuffleboard-api` |
| APP | `shuffleboard-app-all` |

CC @JLLeitschuh

JLLeitschuh, 2017-09-19:
Do you want to publish this as a maven artifact that other devs can pull down from some maven repo?

Do you want to simply have a JAR that devs pull down and use to compile against in their projects?

I would advocate going for the former (with maven), additionally, I would embed a "sample" project in this repository that plugin creators can use as a starting point for creating their own plugins. (All the gradle boilerplate stuff, ect...)

SamCarlberg, 2017-09-19:
I think publishing the jar to the normal FRC maven coordinates (`first.wpi.edu/FRC/roborio/maven`) would be best. The shuffleboard app wouldn't use it, but plugin developers can use it as a compile-time dependency.

I've already put in some work on a skeleton project that's already set up with gradle and the code needed for pushing the jar to the local plugin directory

PR #266 (2017-10-29): Publish API and plugin jars and source and javadocs for all artifacts
Fixes #193
Fixes #265

Plugins are currently distributed as normal jars, (not as uberjars). A good improvement would be to bundle non-shuffleboard dependencies so third-party plugins dependent on the bulitin ones (see #263) are able to use their libraries, eg `ntcore` for a plugin that depends on the NetworkTables plugin. Unfortunately, it looks like it's difficult to exclude transitive dependencies so even if the API files are excluded, all of its dependent libraries (and their dependents, and so on) will be bundled in ever [... cut]
commit 5b264a4e (2017-10-29): Publish API and plugin jars and source and javadocs for all artifacts (#266)

* Publish API and plugin jars and source and javadocs for all artifacts

Fixes #193
Fixes #265

* Fix plugin publishing

Publishes normal jars, but lists dependencies in the POM
Fixed settings.gradle only working for first-level gradle.kts files
closed 2017-10-29 by SamCarlberg