wpilibsuite/shuffleboard #197: Widgets in palette are way too big

bradamiller, 2017-09-21:
The widgets in the palette are huge and should be smaller so you can see more of them at once.

NickBettsRandom, 2018-02-06:
I would second this. Seeing everything when debugging is a real challenge, and that is considering we are usually using second monitor to display the shuffleboard. I was considering trying to make a custom widget just to squeeze more information in.
However, I just discovered that there is a solution using theme style sheets. If you use the following code in a theme it will make the text much smaller for the tiles, allowing them to be made smaller. Note there are other text sizes that can be changed too. See the [Theme Documentation](https://github.com/wpilibsuite/shuffleboard/wiki/Custom-themes) for details on making a custom theme.

`.tile { -fx-font-size: 8pt; }`
`.tile-title-bar { -fx-font-size: 8pt; }`

SamCarlberg, 2018-02-06:
@NickBettsRandom This issue is referring to the previewed widgets in the "Widgets" tab. You can decrease the tile size in the tab preferences, or lower the default value used in new tabs in the application preferences

carbotaniuman, 2020-08-24:
@PeterJohnson Disposition to close.

closed 2020-08-24 by PeterJohnson