wpilibsuite/shuffleboard #214: loadSimpleWidget Test disabled

AustinShalit, 2017-10-02:
Take a look at my attempts to fix this here: https://github.com/wpilibsuite/shuffleboard/pull/204

For some reason, Jenkins refuses to run tests with GSON correctly.  So for now, I added a tag, `NonJenkinsTest`, that we should use for tests that use GSON and fail on Jenkins.  Hopefully, we come across a solution at some point.  This does mean we need to leave Travis CI & AppVeyor as required for build because they will be testing functionality that Jenkins does not.

SamCarlberg, 2018-04-02:
This may be related to #358, which was fixed in #384

carbotaniuman, 2020-09-06:
OBE. Disposition to close @PeterJohnson

AustinShalit, 2020-09-10:
Why is this OBE?

carbotaniuman, 2020-09-10:
We don't use Jenkins or Appveyor or Travis anymore?

AustinShalit, 2020-09-10:
We should probably open a PR that removes the NonJenkins flag from all tests

PR #667 (2020-09-13): Remove OBE annotations
Fixes #214.
closed 2020-09-13 by AustinShalit