wpilibsuite/shuffleboard #221: The label for a widget inside a layout doesn't show with the ... on the left side

bradamiller, 2017-10-04:
The names of widgets placed directly in the grid show the rightmost part of the name if it is too big to fit. The names of widgets inside a list don't do that.

SamCarlberg, 2017-10-04:
I think this is pretty minor:

1. Layouts are in scroll panes, so the text only wraps after ~28 characters (at minimum; this increases with tile width)
2. Subsystem widgets are the major usecase for layouts, and will automatically remove the leading subsystem name from the keys. I don't see component names exceeding that limit (eg `"Front Left Drive CAN Talon"` is 26, with spaces, and that's the longest I can think of)

bradamiller, 2017-10-05:
Actually, as we talked about, I think it would be best to only have the final part of the network table key since the typical use case is doing things like SmartDashboard.putNumber("x", 5). In that case, it should really create a text field with a label of "x" and a value of 5. And leave off the front part of the path completely. I know it doesn't solve the case of putNumber("a/b/c", 4), but that's such a weird case and I think it would be OK just to call it "c".

PR #306 (2017-12-21): General housekeeping cleanup
Fixes lots of minor issues:

- Fixes #44
- Fixes #115
- Fixes #127
- Fixes #142
- Fixes #221
- Fixes #264
- Fixes #272

Also improves performance of previewing tiles from dropped sources by caching the size rather than creating and computing the size of a new widget/layout EVERY time the mouse is moved while dragging
commit ead9bfb8 (2017-12-21): General housekeeping cleanup (#306)

* Rename widget classes to end with "Widget", rename WidgetPropertySheet

Fixes #142
Fixes #264
Fixes #272

* Use ~/Shuffleboard instead of ~/Smartdashboard

Replaces most instances of "Smartdashboard" to "Shuffleboard", except for the Smartdashboard tab since that contains stuff in the Smartdashboard table from the robot
Fixes #44

* Show grid highlight when resizing tiles

Fixes #127

* Preview size of dropped widgets from the gallery

Also improves performance of previewing tiles from dropped sources by caching the size of the widget rather than creating [... cut]
closed 2017-12-21 by bradamiller