wpilibsuite/shuffleboard #264: Widget class names should all end with "widget"

SamCarlberg, 2017-10-28:
This would fix some headaches when trying to use UI controls that have the same or similar name to a widget

| Widget | Confused With | New Name |
|---|---|---|
| `NumberSlider` | `javafx.scene.control.Slider` | `NumberSliderWidget` |
| `ToggleButton` | `javafx.scene.control.ToggleButton` | `ToggleButtonWidget` |
| `ToggleSwitch` | `org.controlsfx.control.ToggleSwitch` | `ToggleSwitchWidget` |
| `TextView` | `TextField`, `TextArea` | `TextViewWidget` |
| `BooleanBox` | N/A | `BooleanBoxWidget` |
| `SpeedController` | N/A | `SpeedControllerWidget` |

AustinShalit, 2017-10-29:
Should widgets be in their own package too?

SamCarlberg, 2017-10-29:
They already are

PR #306 (2017-12-21): General housekeeping cleanup
Fixes lots of minor issues:

- Fixes #44
- Fixes #115
- Fixes #127
- Fixes #142
- Fixes #221
- Fixes #264
- Fixes #272

Also improves performance of previewing tiles from dropped sources by caching the size rather than creating and computing the size of a new widget/layout EVERY time the mouse is moved while dragging
commit ead9bfb8 (2017-12-21): General housekeeping cleanup (#306)

* Rename widget classes to end with "Widget", rename WidgetPropertySheet

Fixes #142
Fixes #264
Fixes #272

* Use ~/Shuffleboard instead of ~/Smartdashboard

Replaces most instances of "Smartdashboard" to "Shuffleboard", except for the Smartdashboard tab since that contains stuff in the Smartdashboard table from the robot
Fixes #44

* Show grid highlight when resizing tiles

Fixes #127

* Preview size of dropped widgets from the gallery

Also improves performance of previewing tiles from dropped sources by caching the size of the widget rather than creating [... cut]
closed 2017-12-21 by bradamiller