wpilibsuite/shuffleboard #293: Add warning to edit grade-wrapper.properties in README

aswerdlow935, 2017-11-17:
Tried to shuffleboard with both Java JDK 8 & 9 installed (at least on a mac), and shuffleboard defaulted to Java JDK 9 even after gradle.property and bash_profile changes. However, adding `javaHome=/Library/Java/JavaVirtualMachines/jdk1.8.0_131.jdk/Contents/Home` fixes the issue. A warning should be added in the README to add this property if running fails with the unknown 9.0.1 error message.

JLLeitschuh, 2017-11-17:
More robust solution, before you run gradle, you just run this:
```bash
JAVA_HOME=`/usr/libexec/java_home -v 1.8`
```
or if you are using the Fish shell:
setenv -gx JAVA_HOME (/usr/libexec/java_home -v 1.8)

aswerdlow935, 2017-11-22:
That also works just as well...thanks👌

closed 2017-11-22 by aswerdlow935