wpilibsuite/shuffleboard #305: No support for arrays

SamCarlberg, 2017-11-21:
The data types exist, but the only array type that can be viewed is `NumberArray`, and only by graphs. There should be a widget for viewing arrays. A list view would probably be enough for read-only arrays.

Supersedes #189

Martysh12, 2024-10-18:
Is this being worked on? I've made a plugin for displaying array types, and I'd like to make a pull request to merge it with Shuffleboard.
[image]

SamCarlberg, 2024-10-18:
> Is this being worked on? I've made a plugin for displaying array types, and I'd like to make a pull request to merge it with Shuffleboard.
> [image]
> 

Nope! Open a PR if you'd like to contribute your work