wpilibsuite/shuffleboard #310: Improper handling of deleted NTCore entries

JacisNonsense, 2017-11-29:
When deleting an entry with NetworkTables (i.e. `table.getEntry("myEntry").delete()`), the key remains in the outline viewer (and the tab, if a widget is present). 

If the widget is not present in the tab, the deleted entry that still remains in the outline viewer cannot be dragged into the tab (i.e. cannot be interacted with). Deleted entries appear to be ghosting.

bradamiller, 2017-11-29:
Thanks for the report.

Just a quick question, did you notice if the OutlineViewer standalone program also has the same issue?

JacisNonsense, 2017-11-29:
OutlineViewer standalone does not have this same issue

SamCarlberg, 2018-01-03:
I am unable to reproduce this when deleting entries through outlineviewer. Deleting an entry through outlineviewer removes the entry from the sources view in shuffleboard.

However, deleting an entry does not delete the cached source or disable existing widgets for that source, allowing the key to be re-added by eg typing into a text field, or throwing an exception if an entry is added with the same key but a different data type