wpilibsuite/shuffleboard #311: NetworkTable Outline View refresh flicker

PeterJohnson, 2017-11-30:
The outline view of the NetworkTable has severe flicker when there are a decent number of entries (e.g. the PacGoat example with multiple subsystems).  I see the outline view itself flicker, and also see the scrollbar resize.  Happens every few seconds, and is very visually distracting.  Most likely this is caused by updating the display while the data in the outline is being updated?

19lmyers, 2018-02-26:
I'd just like to say that I'm seeing this as well on the latest version. It's quite annoying.

SamCarlberg, 2018-04-02:
I've noticed that the flicker gets worse as the application or render threads get bogged down. Very noticeable when many large graphs are visible, but otherwise seems to be very rare. The dark theme gets annoying, though, as the `:hover` property doesn't seem to work correctly and makes other, non-hovered rows highlighted