wpilibsuite/shuffleboard #314: Better support grouping by subsystem?

PeterJohnson, 2017-12-03:
With automatic LiveWindow support, there are a LOT more widgets being added to LiveWindow.  Should there be an ability to group them better by subsystem?  Not sure how this can be done easily in the current graphical structure.

PeterJohnson, 2017-12-03:
Actually, this was an error in wpilibsuite/allwpilib#771 where subsystems weren't getting type=LW Subsystem

closed 2017-12-03 by PeterJohnson

SamCarlberg, 2017-12-03:
All subsystems should just be classified as "Subsystem" anyway, since there's no difference between "Subsystem" and "LW Subsystem", and "PID Subsystem" just has an embedded PIDController that gets recognized anyway.

PeterJohnson, 2017-12-05:
SmartDashboard requires it be "LW Subsystem", and so does Shuffleboard currently (otherwise the subsystem controls are not grouped).

SamCarlberg, 2017-12-05:
SmartDashboard shows "Subsystem" with just a text field for a command to run, while "LW Subsystem" shows all of the components in the subsystem. I guess we can keep it as-is.