wpilibsuite/shuffleboard #340: Attempting to close a tab should ask for confirmation

SamCarlberg, 2018-01-05:
The current behavior makes it easy to accidentally close a tab, with no way to restore it.

This should at the very least ask for confirmation for populated tabs, and should be configurable in the global app settings

Team3840, 2018-02-12:
How do you resolve the issue when the two default tabs are missing and does auto populate correctly?  We have computers that their are no defaulted tabs with "SmartDashboard" and "LiveWindow" , instead is just have blank "tab1"  This forces users to always re-setup the tabs.

[image]

SamCarlberg, 2018-02-12:
File -> New Layout will create a new layout separate from the save file. You'll have to save it over the old file

Team3840, 2018-02-20:
So once the tabs disappear and trying to recreate them to save as new layout it appears to be broken every time it's reopen.  Even if you open saved layout it does not open anything.  Leaving the auto populate not working as well.  This is on the Feb 15 release.  This is a real issue going into the events coming.  Is there any further updates coming?

SamCarlberg, 2018-02-20:
Please share the log file at <home>/Shuffleboard/shuffleboard.log

Team3840, 2018-02-20:
[shuffleboard.log](https://github.com/wpilibsuite/shuffleboard/files/1740655/shuffleboard.log)

Team3840, 2018-02-20:
[shuffleboard json.zip](https://github.com/wpilibsuite/shuffleboard/files/1740661/shuffleboard.json.zip)

SamCarlberg, 2018-02-20:
The save file has a tab with a grid size of 25, when the minimum value is 32 (widgets tend not to display correctly under this size). This is preventing the save file from being loaded. You can edit the save file and replace the "gridSize": "25" line with "gridSize": "32", or delete the entire block for that tab (since it's empty), or just create a new layout and save over the old one

Team3840, 2018-02-20:
[image]

In the above postings I have the log file and the saved layout.  The saved layout does load either.

Team3840, 2018-02-20:
Perfect that works.  So in future releases the grid size should have minimum value limit check for preventing this from happening.  I tweak the layout file in Notepad++ for gridSize and reopened in shuffle board just fine.  I also set my default grid size to 50 now.  Thanks.

PR #542 (2018-11-12): Request user confirmation when closing tabs
The confirmation behavior is active by default and can be disabled in the app settings

Closes #340 
Closes #541
closed 2018-11-12 by SamCarlberg