wpilibsuite/shuffleboard #342: Shuffleboard plugin loading error, or documentation error?

19lmyers, 2018-01-05:
I created a test Shuffleboard plugin, but I can't seem to get it to load.

I instead get this error in the console:
```
java.lang.NoSuchMethodError: edu.wpi.first.shuffleboard.api.plugin.Plugin: method <init>()V not found
        at com.team980.shuffleboard.theme.ShuffleboardTheme.<init>(ShuffleboardTheme.java:16)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(Unknown Source)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
        at java.lang.reflect.Constructor.newInstance(Unknown Source)
        at java.lang.Class.newInstance(Unknown Source)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.loadPluginClass(PluginLoader.java:124)
        at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
        at java.util.stream.Streams$StreamBuilderImpl.forEachRemaining(Unknown Source)
        at java.util.stream.ReferencePipeline$Head.forEach(Unknown Source)
        at java.util.stream.ReferencePipeline$7$1.accept(Unknown Source)
        at java.util.stream.ReferencePipeline$3$1.accept(Unknown Source)
        at java.util.ArrayList.forEach(Unknown Source)
        at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
        at java.util.stream.Sink$ChainedReference.end(Unknown Source)
        at java.util.stream.Sink$ChainedReference.end(Unknown Source)
        at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
        at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
        at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
        at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
        at java.util.stream.ReferencePipeline.forEach(Unknown Source)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.loadPluginJar(PluginLoader.java:98)
        at edu.wpi.first.shuffleboard.app.PluginPaneController.lambda$loadPlugin$7(PluginPaneController.java:106)
        at java.util.ArrayList.forEach(Unknown Source)
        at java.util.Collections$UnmodifiableCollection.forEach(Unknown Source)
        at edu.wpi.first.shuffleboard.app.PluginPaneController.loadPlugin(PluginPaneController.java:103)
        ... 58 more
```

This is my plugin:
```
package com.team980.shuffleboard.theme;

import com.google.common.collect.ImmutableList;
import edu.wpi.first.shuffleboard.api.plugin.Description;
import edu.wpi.first.shuffleboard.api.plugin.Plugin;
import edu.wpi.first.shuffleboard.api.theme.Theme;

import java.util.List;

@Description(
        group = "com.team980.shuffleboard",
        name = "ShuffleboardTheme",
        version = "1.0.0",
        summary = "Shuffleboard theme for FRC Team 980 ThunderBots"
)
public class ShuffleboardTheme extends Plugin {

    public static final Theme THUNDER = new Theme("Thunder", "/theme.css");

    @Override
    public List<Theme> getThemes() {
        return ImmutableList.of(THUNDER);
    }
}
```

I'm not sure what the problem is, namely because it could be a Shuffleboard error when loading the plugin or the documentation may be wrong about what's required in a Plugin subclass. I did also try adding an empty constructor and a constructor that calls super(), but those resulted in the same issue.

SamCarlberg, 2018-01-05:
What does your build file look like? Your plugin class is fine

19lmyers, 2018-01-05:
Build.gradle:
```
group 'com.team980.shuffleboard'
version '1.0.0'

apply plugin: 'java'

sourceCompatibility = 1.8

repositories {
    mavenCentral()
    maven {
        url 'http://first.wpi.edu/FRC/roborio/maven/release/'
    }
}

dependencies {
    implementation group: 'edu.wpi.first.shuffleboard', name: 'api', version: '1.0.0'

    testCompile group: 'junit', name: 'junit', version: '4.12'
}
```

SamCarlberg, 2018-01-05:
Okay, your build script looks good as well.

I've created an example plugin repo [here](https://github.com/SamCarlberg/example-shuffleboard-plugin) and a release jar [here](https://github.com/SamCarlberg/example-shuffleboard-plugin/releases/tag/1.0-SNAPSHOT). Can you confirm that the example jar loads for you? If it doesn't, my only other suggestion would be to reinstall shuffleboard (can download the jar [here](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/shuffleboard/app/1.0.0/app-1.0.0.jar)) and try that

19lmyers, 2018-01-05:
That makes sense - I was running Shuffleboard through GradleRIO, and it reports an older version. The plugin loads fine with the release version you linked.

closed 2018-01-05 by 19lmyers