wpilibsuite/shuffleboard #346: Jar File does not Launch

JDaniel29, 2018-01-05:
Whenever I try to launch the jar file bundled in version ```2018.1.1``` of the wpilib suite and the current version of this repository, I get this error in the Windows command line:
```
Jan 05, 2018 2:15:08 PM edu.wpi.first.shuffleboard.app.Shuffleboard setupLoggers
CONFIG: Configuration done.
Jan 05, 2018 2:15:09 PM edu.wpi.first.shuffleboard.app.Shuffleboard setupLoggers
CONFIG: Shuffleboard app version: null
Exception in Application start method
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.base/java.lang.reflect.Method.invoke(Unknown Source)
        at java.base/sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
Caused by: java.lang.RuntimeException: Exception in Application start method
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication1(Unknown Source)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication$2(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NoSuchMethodError: com.sun.javafx.tk.FontLoader.computeStringWidth(Ljava/lang/String;Ljavafx/scene/text/Font;)F
        at edu.wpi.first.shuffleboard.api.components.AutoSizedTextField.lambda$new$8(AutoSizedTextField.java:22)
        at org.fxmisc.easybind.EasyBind$4.computeValue(EasyBind.java:101)
        at javafx.base/javafx.beans.binding.ObjectBinding.get(Unknown Source)
        at javafx.base/javafx.beans.binding.ObjectExpression.getValue(Unknown Source)
        at javafx.base/com.sun.javafx.binding.ExpressionHelper.addListener(Unknown Source)
        at javafx.base/javafx.beans.binding.ObjectBinding.addListener(Unknown Source)
        at javafx.base/javafx.beans.binding.DoubleBinding.bind(Unknown Source)
        at javafx.base/javafx.beans.property.DoublePropertyBase$ValueWrapper.<init>(Unknown Source)
        at javafx.base/javafx.beans.property.DoublePropertyBase$2.<init>(Unknown Source)
        at javafx.base/javafx.beans.property.DoublePropertyBase.bind(Unknown Source)
        at javafx.graphics/javafx.css.StyleableDoubleProperty.bind(Unknown Source)
        at edu.wpi.first.shuffleboard.api.components.AutoSizedTextField.<init>(AutoSizedTextField.java:20)
        at edu.wpi.first.shuffleboard.api.components.EditableLabel.<init>(EditableLabel.java:28)
        at edu.wpi.first.shuffleboard.app.components.TabHandle.<init>(TabHandle.java:58)
        at edu.wpi.first.shuffleboard.app.components.DashboardTab.<init>(DashboardTab.java:88)
        at edu.wpi.first.shuffleboard.app.components.DashboardTabPane.createAutoPopulateTab(DashboardTabPane.java:30)
        at edu.wpi.first.shuffleboard.app.components.DashboardTabPane.<init>(DashboardTabPane.java:25)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(Unknown Source)
        at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(Unknown Source)
        at java.base/java.lang.reflect.Constructor.newInstance(Unknown Source)
        at java.base/java.lang.Class.newInstance(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader$InstanceDeclarationElement.constructValue(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader$ValueElement.processStartElement(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader.processStartElement(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader.loadImpl(Unknown Source)
        at javafx.fxml/javafx.fxml.FXMLLoader.load(Unknown Source)
        at edu.wpi.first.shuffleboard.app.Shuffleboard.start(Shuffleboard.java:72)
        at javafx.graphics/com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$9(Unknown Source)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runAndWait$11(Unknown Source)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$9(Unknown Source)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at javafx.graphics/com.sun.javafx.application.PlatformImpl.lambda$runLater$10(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.InvokeLaterDispatcher$Future.run(Unknown Source)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at javafx.graphics/com.sun.glass.ui.win.WinApplication.lambda$runLoop$3(Unknown Source)
        ... 1 more
```

333fred, 2018-01-05:
What version of the JDK are you running? That looks like you're possibly running an outdated version of 8.

JDaniel29, 2018-01-05:
I am running `1.8.0_65`.

333fred, 2018-01-05:
That's way out of date, java 8 is up to update 151. Try updating your JDK/JRE and try again.

SamCarlberg, 2018-01-05:
Shuffleboard does not work with Java 9.

closed 2018-01-05 by SamCarlberg