wpilibsuite/shuffleboard #347: [DOCUMENTATION] What's the root of the path for theme stylesheets?

19lmyers, 2018-01-05:
I'm attempting to make a plugin that registers a custom theme stylesheet, but I can't seem to figure out how to specify the stylesheet path in relation to its location in the .jar.

Example: I place my custom theme.css at /resources/theme.css and specify the path as "/theme.css", but I get the error `WARNING: Resource "/theme.css" not found.` on theme load.

I tried this without the leading slash, and at the path /resources/com/team980/shuffleboard/theme/theme.css and "/com/team980/shuffleboard/theme/theme.css", but I still got the same error.

SamCarlberg, 2018-01-05:
It should be an absolute path. So if the css file is in src/main/resources/theme.css, the path you give should be "/theme.css". If it's in src/main/resources/foo/bar/theme.css, the path should be "/foo/bar/theme.css"

19lmyers, 2018-01-05:
I'm giving the path "/theme.css" and it's located at /src/main/resources/theme.css, same error

SamCarlberg, 2018-01-05:
Ah, this is definitely a bug. No resources get extracted from plugin JARs, so custom themes or FXML widgets or anything else that depends on non-class files will not be usable

SamCarlberg, 2018-01-08:
@19lmyers #349 fixes this. The theme API has changed slightly for plugin authors; you'll need to provide some class that's defined in your plugin (it could be the plugin class itself, or something else) in order for shuffleboard to be able to discover the stylesheets.

But if you're only using the plugin API to define a theme, you'll now be able to place the stylesheets in the Shuffleboard directory to define the theme without having to write any code. Look at PR for details on how this is done

19lmyers, 2018-01-08:
I'll look into that. Thanks

commit 03204bb3 (2018-01-15): Allow themes to be defined in the Shuffleboard directory (#349)

This makes it so that themes can be easily defined and edited without having to write any code

Fixes #347
closed 2018-01-15 by SamCarlberg