wpilibsuite/shuffleboard #352: Minor regression in MJPG Stream Viewer

bhjelstrom, 2018-01-11:
We post the limelight camera stream to NT such that it is visible to all dashboards (<ip>:5800). Shuffleboard, however, is the only dashboard that could not view the stream before we fixed this bug in our code:
In our initial connection header, we were including the "--" before the boundary name. 
I think this is technically our fault, but no other mjpg have had any trouble with the extra "--" included before the boundary name.

PeterJohnson, 2018-01-11:
Yep, that's definitely not compliant with the MIME multipart/x-mixed-replace standard (https://www.w3.org/Protocols/rfc1341/7_2_Multipart.html).  It looks like some other mjpg readers simply ignore the boundary setting in the initial connection header and take whatever comes after the first "--" as the boundary.

ThadHouse, 2018-02-06:
This is actually a cscore issue, not a shuffleboard issue. We should move this to that repo.

SamCarlberg, 2018-02-12:
This appears to have been fixed in a recent limelight update

closed 2018-02-12 by SamCarlberg