wpilibsuite/shuffleboard #353: MJPG Stream Viewer performance

bhjelstrom, 2018-01-11:
See shuffleboard, chrome, and smartdashboard connect to the Limelight camera stream here:
https://gfycat.com/CapitalEmbellishedBoutu

Shuffleboard is skipping frames, and frames even appear to be compressed on arrival in the shuffleboard viewer.

SamCarlberg, 2018-01-11:
Can you run the shuffleboard jar from the command line with

`java -jar -Djavafx.animation.fullspeed=true Shuffleboard-v1.0.0.jar`

This might be due to JavaFX's default tick rate of 60FPS having issues with getting images at 90FPS.

bhjelstrom, 2018-01-11:
I'll try that soon, but the camera stream from Limelight is actually only 15 fps at the moment. 

Another hint is that the stream looks better as I increase the framerate. What was shown in the gif isn't a maximum framerate.

bhjelstrom, 2018-01-12:
No change with the fullspeed option
[image]

bhjelstrom, 2018-01-15:
There isn't a fix for shuffleboard perf yet, but we did fix a Limelight-side issue that prevented the stream from working at all. (https://github.com/wpilibsuite/shuffleboard/issues/352)

SamCarlberg, 2018-01-15:
The only other shuffleboard-side explanation I can think of is that the image data isn't available to the JavaFX thread immediately, which would have been fixed by #345 by making the data property update atomically rather than from the JavaFX thread. @bhjelstrom can you run shuffleboard from the current master branch and see if the performance hit is still present?

bhjelstrom, 2018-01-19:
No change with the latest master...

ThadHouse, 2018-02-06:
I was able to reproduce this completely without shuffleboard using only cscore. So we should move this, along with #352 over to the cscore repo.

SamCarlberg, 2018-02-12:
Fixed by https://github.com/wpilibsuite/cscore/pull/119

closed 2018-02-12 by SamCarlberg