wpilibsuite/shuffleboard #358: Layout is not saved to disk

rogerfachini, 2018-01-13:
When attempting to save the Shuffleboard layout to disk, it writes a .json file 0KB in size, and the following exception is logged: 

 `java.lang.IllegalArgumentException: class com.sun.javafx.jmx.HighlightRegion declares multiple JSON fields named hash`. 

I have tried saving in multiple different folders (all of which my account has write access to), and trying different combinations of dashboard widgets.  The dashboard had 9 Boolean Box widgets, one Command widget, one Dropdown widget, one PowerDistributionPanel widget, and one String widget. 

shuffleboard.log: https://pastebin.com/gX9uYDSJ

Operating System: Windows 10 1709
Java version:  "1.8.0_151"
Java(TM) SE Runtime Environment (build 1.8.0_151-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.151-b12, mixed mode)

SamCarlberg, 2018-01-15:
I remember occasionally seeing this in testing. Looks like Gson is freelancing and not using our serialization classes. @nightpool do you remember what the cause was?

SamCarlberg, 2018-01-27:
@rogerfachini can you please test #384 and see if it fixes the problem?

gbui, 2018-01-30:
@SamCarlberg I had the same issue and it works great!

SamCarlberg, 2018-01-30:
Merged 384 since @gbui has confirmed it works for them. @rogerfachini if the problem persists, feel free to re-open this issue

closed 2018-01-30 by SamCarlberg

SamCarlberg, 2018-05-23:
I usually see that happen when GSON attempts to serialize a JavaFX property whose bean has been set to something that contains a reference to a object in the scene graph. We got around that by explicitly serializing the name and value of the property to the JsonObject eg 

```js
JsonObject object = ...
Property<?> property = ...
object.add(property.getName(), context.serialize(property.getValue());
```

c0ver, 2018-05-24:
Thanks for the help. Apparently, it was a `javafx.scene.control.Button` that was causing errors for some reason. I just made GSON ignore it with the `transient` keyword.