wpilibsuite/shuffleboard #361: Shuffleboard segfaults on startup on some setups

blucoat, 2018-01-14:
When I start shuffleboard on my machine, I get the following message at startup:

```
10:56 $ java -jar Shuffleboard.jar 
Gtk-Message: Failed to load module "atk-bridge"
Jan 12, 2018 10:56:45 AM java.util.logging.LogManager$RootLogger log
CONFIG: Configuration done.
Jan 12, 2018 10:56:45 AM java.util.logging.LogManager$RootLogger log
CONFIG: Shuffleboard app version: 1.0.0
Jan 12, 2018 10:56:45 AM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:Base:1.0.0
Jan 12, 2018 10:56:46 AM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 8.0.121 by JavaFX runtime of version 8.0.111
Jan 12, 2018 10:56:46 AM sun.awt.X11.XToolkit setBackingStoreType
CONFIG: The system property sun.awt.backingStore is not set, by default backingStore=NotUseful
Jan 12, 2018 10:56:47 AM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 8.0.121 by JavaFX runtime of version 8.0.111
Jan 12, 2018 10:56:47 AM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:NetworkTables:1.0.0
Jan 12, 2018 10:56:47 AM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 8.0.121 by JavaFX runtime of version 8.0.111
Jan 12, 2018 10:56:47 AM javafx.fxml.FXMLLoader$ValueElement processValue
WARNING: Loading FXML document with JavaFX API of version 8.0.121 by JavaFX runtime of version 8.0.111
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f34997848c9, pid=9546, tid=0x00007f3468d69700
#
# JRE version: OpenJDK Runtime Environment (8.0_151-b12) (build 1.8.0_151-8u151-b12-0ubuntu0.17.04.2-b12)
# Java VM: OpenJDK 64-Bit Server VM (25.151-b12 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libLLVM-4.0.so.1+0x6e18c9]  llvm::sys::path::reverse_iterator::operator++()+0xc9
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/james/wpilib/tools/hs_err_pid9546.log
Jan 12, 2018 10:56:47 AM edu.wpi.first.shuffleboard.app.plugin.PluginLoader load
INFO: Loading plugin edu.wpi.first.shuffleboard:CameraServer:1.0.0
Jan 12, 2018 10:56:47 AM edu.wpi.first.shuffleboard.plugin.cameraserver.CameraServerPlugin onLoad
INFO: OpenCV version: 3.2.0
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
Aborted (core dumped)
```

After spending the last few days screwing around with gdb, I discovered that the segfault was coming from when networktables tries to connect to localhost (and fails), and then logs the error.  The `LoggerImpl::Log` function calls `llvm::sys::path::filename` to get the name of the source file where the error happened.  However, there are two versions of this function in the address space at runtime: one is included in wpiutil, which is statically linked into `libntcore.so`, and the other is in the system `libLLVM-4.0.so`.  The system libLLVM is getting pulled in by `radeonsi_dri.so`, which is part of Mesa (the open-source OpenGL implementation).  The problem is that the logger function ends up calling the system libLLVM when it meant to call the one in wpiutil.  There must be some slight difference in how these work, because it ends up causing a segfault (from poking around with gdb, my guess is that it has to do with how the `StringRef` class uses the MSB of the size as a flag.  When the segfault happens, the index into a string is almost correct, except it has the MSB set.)

I tried messing with compiler flags to get this to work (`-fno-semantic-interposition` and `-fno-plt`) with no success, but I'm not very knowledgeable about how interposition works.

Changing the namespace `llvm` to `wpi_llvm` in wpiutil and networktables fixed it for me.  If we're okay with renaming it and making downstream projects deal with it (Do teams end up using these functions in robot code?) then I can make a PR on those projects.

PeterJohnson, 2018-01-14:
Yes, these functions are used extensively in robot code (namely llvm::StringRef and llvm::Twine).  We'll look at moving the imported llvm stuff into the wpi namespace for next year, but changing this mid-season is far too major of a change.

AustinShalit, 2018-05-30:
Closed by https://github.com/wpilibsuite/allwpilib/pull/1006

closed 2018-05-30 by AustinShalit