wpilibsuite/shuffleboard #367: UI Elements not Rendering

JohnGilb, 2018-01-17:
When loading Shuffleboard, a lot of the UI is failing to render - menus are empty, dialog windows are blank...

[image]

SamCarlberg, 2018-01-17:
Operating system? Java version and vendor?

JohnGilb, 2018-01-19:
Sorry for the delay - needed to get back to the system with the issue.

Windows Version:   10.0.16299 Build 16299
Java: JDK/JRE 1.8.0_151 from Oracle

nollchr, 2018-01-20:
Same issue on windows 10.  Slack desktop application also rendering completely white.  Updated to latest NVIDIA device drivers (Quadro M1000 GPU) and that didn't fix it.

Windows 10 Pro.
Java:  JDK/JRE 1.8.0_75

ianjessett, 2018-04-14:
Hi Guys, did you reach a resolution on this one? We have just installed to a new driver station laptop and we are seeing exactly the same symptoms - a blank shuffleboard. Ours is Windows 10 Pro with JDK 1.8.0_162. Tried the shuffleboard from the eclipse WPILIB Robot Development plugins (version 2018.4.1) and also the newer version from http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/shuffleboard/app/1.3.1/app-1.3.1.jar
Same problem in both unfortunately.

JohnGilb, 2018-04-14:
Sorry to not mention this earlier: I resolved my issue by downloading/installing the newest graphics card drivers for my computer.

ianjessett, 2018-04-14:
Thanks for the hint. I have found a newer driver for the nvidia display adaptor. Unfortunately that did not solve the problem for us. What did work - changing the graphics processor to "integrated graphics" for java.exe in the NVidia control panel - we now see content in the shuffleboard. Not ideal but it points to the nvidia driver or it's configuration being the issue in our case.

SamCarlberg, 2018-08-16:
Closing this as it's an issue with your graphics drivers and not something we can fix

closed 2018-08-16 by SamCarlberg