wpilibsuite/shuffleboard #374: Question: Custom Widgets

pjreiniger, 2018-01-20:
TL;DR - A good end-to-end (robot publishing a NT value -> custom widget displaying the information) example would be helpful for teams trying to create custom widgets.

I'm trying to get spun up making a custom widget for this, and I'm not totally understanding the documentation.  I think a lot of the logic is abstracted away in a manner such that a lot of things happen behind the scenes with magic and hand waving.  The documentation shows very sparse shells and none seem to have the full pipeline illustrated.  I've been trying to figure out what is going on, but in the meantime I though asking these questions might work out 100x faster.

My team has made [custom SD widgets](https://github.com/ArcticWarriors/snobot-2017/blob/submodules/SmartDashboardExtensions/2017SmartDashboardExtensions/src/com/snobot/sd2017/robot/RobotWidget.java) the past couple of years.  I think I originally based them off of an example that was given, or was able to easily follow what one of the standard widgets did.  Because of the SD's simplicity and stupidity, the steps were pretty straight forward:

1. Create a class that extends StaticWidget
2. Create a JPanel or whatever to put on there
3. Add listeners to the Robot.getTable() table, or run a polling timer, and read Robot.getTable().getDouble("myKey") (or whatever)

Sometimes we would put things in subtables and pull the data out of there, but because it was a manual process everything was very clear.  It is very probable that we were not doing things the "recommended" or best way, but it was easy to follow.  I think the thing I am most confused about is how data gets pulled from the NetworkTables and populates javafx panes.

Furthermore, what is the recommended way to package the data in the NT?  Make a subtable for our "robot widget", "coordinate widget", and "autonomous file widget"?  Or can we still use the SD table for that stuff?  Do I need to make DataType things for these widgets?  

I think the delayed update publications is further confusing me, or at least making it more annoying for me to test with.  I've been following [this issue](https://www.chiefdelphi.com/forums/showthread.php?p=1723401) and its fix.  I wish he went that final step and got data into his widget, that might have answered all the questions I had.  In either case, lets get #371 in there.

SamCarlberg, 2018-01-20:
Have you read the wiki pages on widget and plugin development?

[Defining a widget](https://github.com/wpilibsuite/shuffleboard/wiki/Defining-a-widget)
[Widget types](https://github.com/wpilibsuite/shuffleboard/wiki/Widget-types)
[Defining custom data types](https://github.com/wpilibsuite/shuffleboard/wiki/Defining-a-new-data-type)
[Plugins](https://github.com/wpilibsuite/shuffleboard/wiki/Plugins)

But now that you've mentioned it, we really are lacking documentation for an end-to-end example for creating widgets for custom data types.

---

If you're sending basic data (primitives or sendables defined in WPILib), you won't need to deal with metadata. Single entries are treated as independent, and the type is determined purely by the type of the entry's value. For sendables, they already define the `.type` metadata entry.

However, if you want to send something more complicated (say, a point for the robot's coordinate, or location + angle, or...), then you'll need to either define a sendable data type that sets up the table and updates its entries, or you'll have to manually update the entries yourself

---

The general structure is

```
NetworkTables <-> NetworkTable sources <-> Widgets
```

Widgets have _zero_ access to the underlying data stream. This lets the same widget be used to show the same data type from different source types; a Toggle Button widget can interact with a boolean value, regardless if it comes from NetworkTables, a data recording, or a custom data source like ZeroMQ or HTTP/REST.

This is very different from SmartDashboard, since it removes access to NetworkTables from the widget; everything is brokered by the data source API.

---

If you're sending a custom data type through NetworkTables, you need to add a metadata entry `".type"="MyDataType"`. This is the entry that the NetworkTable source will look for to determine the data type. If that entry isn't present, it'll treat it as a generic "map" type and display the data in a table.

So in the `Point` data type example in the custom data types page, you would structure the subtable as

```
/<table>
  - .type = "Point"
  - x = 2.17
  - y = 3.14
```

There also needs to be a registered data type for the Point data so that NetworkTables knows how to interpret the table.

Continuing the `Point` example, the simplest widget for it would look something like

```java
@Description(name = "Point Widget", dataTypes = Point.class)
@ParametrizedController("PointWidget.fxml")
public class PointWidget extends SimpleAnnotatedWidget<Point> {

  @FXML
  private Pane root;
  @FXML
  private Label display;

  @FXML
  private void initialize() {
    display.textProperty().bind(
        dataOrDefault.map(point -> "X: " + point.getX() + " Y: " + point.getY()));
  }

  @Override
  public Pane getView() {
    return root;
  }
}
```

and your view would be in a FXML file looking something like

```xml
<StackPane xmlns="http://javafx.com/javafx/8.0.111" xmlns:fx="http://javafx.com/fxml/1"
           fx:id="root"
           fx:controller="your.package.name.PointWidget">
  <Label fx:id="display"/>
</StackPane>
```

pjreiniger, 2018-01-21:
That was perfect.  I assumed that that bind stuff handled all the black magic going on in the background (I should learn more about javafx).  I got an example working using the complex data example, which should fulfill our needs for the widgets we typically make.  

I'll leave this to you to close, if you want to update the docs with the stuff you put in here before this is "finished", but I'm all set.  At least minus #377

scarmain, 2018-01-22:
Small hijack of the thread...

On https://github.com/wpilibsuite/shuffleboard/wiki/Plugins#creating-a-custom-plugin, there is a link to the Javadoc to know what methods to implement: http://first.wpi.edu/FRC/roborio/release/docs/shuffleboard/api/edu/wpi/first/shuffleboard/api/plugin/Plugin.html

That link 404s, can we get that fixed?

SamCarlberg, 2018-01-29:
The docs will be published in the next WPILib release

closed 2018-03-14 by SamCarlberg