wpilibsuite/shuffleboard #376: Custom Themes?

JarrettFritz, 2018-01-20:
I am trying to find a way to creating a custom theme, and I noticed there is little documentation on how to get there. All I know it is a css file and that it goes into your ShuffleBoard/themes/ folder. Does anyone know any better documentation or how to find the light.css or dark.css file? I cannot find it anywhere.
Thanks!

SamCarlberg, 2018-01-20:
I'm working on a wiki page for this. Should be up in the next few hours

SamCarlberg, 2018-01-20:
It's up: https://github.com/wpilibsuite/shuffleboard/wiki/Custom-themes. Feel free to ask if you any questions

closed 2018-01-20 by SamCarlberg

JarrettFritz, 2018-01-21:
Is there a way to see an example scene in documentation or is it more of a guess and check method?

SamCarlberg, 2018-01-21:
Like a screenshot?

JarrettFritz, 2018-01-21:
like the css behind it

SamCarlberg, 2018-01-21:
I'm not sure I follow. What do you mean by "scene"?

JarrettFritz, 2018-01-21:
Sorry, I meant seeing an example file for css.

JarrettFritz, 2018-01-21:
found it, i guess im blind. Sorry for wasting your time!

SamCarlberg, 2018-01-21:
Yep, they're all the way at the bottom

TalebHirani, 2018-01-21:
I have my file located in ~/Shuffleboard/themes/darkTheme/darkTheme.css but I am unable to select the theme in the drop down menu. This is what is currently in the .css file:

```
@import "/edu/wpi/first/shuffleboard/api/material.css"; /* Material design CSS for JavaFX components */
@import "/edu/wpi/first/shuffleboard/api/base.css";  /* Material design CSS for shuffleboard components */
@import "/edu/wpi/first/shuffleboard/app/dark.css";  /* CSS for the Material Dark theme */

.root {
-fx-font-family: "BebasNeue";
}

.root {
    -swatch-100: #BBDEFB;
    -swatch-200: #90CAF9;
    -swatch-300: #64BEF6;
    -swatch-400: #42A5F5;
    -swatch-500: #2196F3;
}

```
Any idea as to why I cannot select the theme? I am on MacOS if that would have an impact. Thanks!

SamCarlberg, 2018-01-21:
Are you using the most recent version? There was a release yesterday that enabled this feature

TalebHirani, 2018-01-21:
Ah, yeah, that would do it. I was running an outdated version of it, thanks!