wpilibsuite/shuffleboard #377: Custom plugins don't persist/load across resets

pjreiniger, 2018-01-21:
I made a custom widget (inside of a custom plugin) and got it to work in shuffleboard.  However, when I shut it down and save the config, the next time I load it up it complains that it can't find "MyCustomWidgets" setup information.  It appears the config file saves the widget I added, but not the custom plugin I loaded.

SamCarlberg, 2018-01-21:
How did you load the plugin? If it's in a JAR in ~/Shuffleboard/plugins, it should be automatically loaded at startup (even if it wasn't loaded at the last shutdown -- this has been on my plate for a while).

If you loaded the plugin JAR from the "Load plugin" button on the plugins pane, shuffleboard doesn't remember at shutdown. This has also been on my plate for a while.

pjreiniger, 2018-02-04:
Forgot to respond to this.  Originally I was using the "load plugin" button, but after you response switched to putting the jar where it really should be and it fixed my problem.  You seem to have the button issue tracked in #395, so this is good to close

PR #395 (2018-02-09): Remember loaded plugin JARs between restarts
All plugins in the cache will be attempted to be loaded at startup
Adds a button to the plugin pane to clear the cache
Also fixes checkstyle to work with do-while loops

Fixes #377
closed 2018-02-04 by pjreiniger