wpilibsuite/shuffleboard #388: LiveWindow does not autopopulate

msoucy, 2018-01-27:
After updating, we no longer see new items appear on the SmartDashboard and LiveWindow panels - in fact, those panels aren't even being created.

Versions seen in: 1.1.0 - 1.1.1
Tested on: Windows 7, Windows 10
Java 1.8.0_152 on all machines

SamCarlberg, 2018-01-27:
Please post the save file

msoucy, 2018-01-27:
[shuffleboard.log](https://github.com/wpilibsuite/shuffleboard/files/1670439/shuffleboard.log)

SamCarlberg, 2018-01-27:
Looks like something weird with NetworkTables. If you open up OutlineViewer and connect to the robot, you can save that state and post it here

msoucy, 2018-01-27:
[networktables.ini.txt](https://github.com/wpilibsuite/shuffleboard/files/1670459/networktables.ini.txt)

Github doesn't seem to like uploading ini files...

SamCarlberg, 2018-01-28:
@sciencewhiz how did you get that to happen? I've only made it happen by intentionally deleting the tab definitions from save files

msoucy, 2018-01-28:
Yes, it looks exactly like that

sciencewhiz, 2018-01-28:
I don't have a save file.

sciencewhiz, 2018-01-28:
When I create a save file, it doesn't happen. Probably needs to see if there isn't a save file and create the default tabs.

SamCarlberg, 2018-01-28:
That's exactly what it does already:

```java
Platform.runLater(() -> {
  if (AppPreferences.getInstance().isAutoLoadLastSaveFile()) {
    mainWindowController.load(AppPreferences.getInstance().getSaveFile());
  } else {
    mainWindowController.newLayout();
  }
});
```

sciencewhiz, 2018-01-28:
There should be validation that the file exists and is valid.

`Jan 27, 2018 7:40:28 PM edu.wpi.first.shuffleboard.app.MainWindowController load
WARNING: Couldn't load
java.io.FileNotFoundException: C:\Users\Joe\Shuffleboard\shuffleboard.json (The system cannot find the file specified)
        at java.io.FileInputStream.open0(Native Method)
        at java.io.FileInputStream.open(FileInputStream.java:195)
        at java.io.FileInputStream.<init>(FileInputStream.java:138)
        at com.google.common.io.Files.newReader(Files.java:84)
        at edu.wpi.first.shuffleboard.app.MainWindowController.load(MainWindowController.java:460)
        at edu.wpi.first.shuffleboard.app.Shuffleboard.lambda$start$3(Shuffleboard.java:95)
        at com.sun.javafx.application.PlatformImpl.lambda$null$173(PlatformImpl.java:295)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$174(PlatformImpl.java:294)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$null$148(WinApplication.java:191)
        at java.lang.Thread.run(Thread.java:745)`

msoucy, 2018-01-28:
A little late, but I can confirm that if I disable "load last save file", the two default windows come back

PR #390 (2018-01-29): Fix no tabs present if the last save file was renamed, moved, or deleted
Fixes #388. Enabling "load last save" and deleting/moving/renaming the last save file before opening shuffleboard would cause no tabs to be present.
commit 4e03e145 (2018-01-29): Fix no tabs present if the last save file was renamed, moved, or deleted (#390)

Fixes #388
closed 2018-01-29 by SamCarlberg