wpilibsuite/shuffleboard #396: Dashboard Lockup on Robot Code Restart

Kython89, 2018-02-02:
Whenever we redeploy code to the RoboRio our dashboard locks up.  Part of the NetworkTables table viewer turns black and a majority of our values turn grey.  The only way to restore a functioning dashboard is to restart the dashboard.  We've attached a log and screenshot of this behavior.

[image]
[shuffleboard.log](https://github.com/wpilibsuite/shuffleboard/files/1690959/shuffleboard.log)

GaganBhat, 2018-02-05:
We have been able to replicate this. The exact same problem happens to us. The problem does not occur on smartDashboard though.

SamCarlberg, 2018-02-05:
Looks like a duplicate of #391. Graph widgets aren't being properly reset

closed 2018-02-05 by SamCarlberg

Kython89, 2018-02-05:
@SamCarlberg Are you sure this is a duplicate of #391?  Our dashboard locks up and parts of it become unresponsive.  In #391 they state that their's crashes.

SamCarlberg, 2018-02-05:
Different symptoms, but the same cause.