wpilibsuite/shuffleboard #398: Using 3rd Party Libraries

pjreiniger, 2018-02-04:
TL;DR:
1. Is there a recommendation on how to add 3rd party dependencies to shuffleboard's classpath?
2. Is log4 probably just a one-off error causer because of some weird classpath conflict, or is this a potentially systemic problem?

We have a piece of shared code that we can use in a standalone fashion, or pull out pieces of it into a widget.  After discovering PMD (thanks to this/allwpilib) we changed all of our prints to use log4j.  Because this is in a dependency, we must somehow include those .class files/jars in the classpath.

I (through crossing my fingers and getting lucky) found that if I place log4j.jar into the `extension` directory of the smartdashboard with our plugin jar, it works.  However, with shuffleboard I cannot simply do that.  I also experimented with making a `shadowJar`, and manually making a "fat jar".  In either case I get the following error:

```
NT: connect() to localhost port 1735 timed out
Feb 03, 2018 10:45:11 PM edu.wpi.first.shuffleboard.app.plugin.PluginLoader loadPluginJar
INFO: Attempting to load plugin jar: file:///C:/Users/PJ/Shuffleboard/plugins/2018ShuffleboardWidgets-fat.jar
Exception in Application start method
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
        at java.lang.reflect.Method.invoke(Unknown Source)
        at sun.launcher.LauncherHelper$FXHelper.main(Unknown Source)
Caused by: java.lang.RuntimeException: Exception in Application start method
        at com.sun.javafx.application.LauncherImpl.launchApplication1(LauncherImpl.java:917)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication$154(LauncherImpl.java:182)
        at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.NoClassDefFoundError: javax/jms/MessageListener
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at java.lang.Class.forName0(Native Method)
        at java.lang.Class.forName(Unknown Source)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.tryLoadClass(PluginLoader.java:203)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.lambda$loadPluginJar$6(PluginLoader.java:155)
        at java.util.stream.ReferencePipeline$7$1.accept(Unknown Source)
        at java.util.stream.ReferencePipeline$3$1.accept(Unknown Source)
        at java.util.stream.ReferencePipeline$3$1.accept(Unknown Source)
        at java.util.stream.ReferencePipeline$2$1.accept(Unknown Source)
        at java.util.Iterator.forEachRemaining(Unknown Source)
        at java.util.Spliterators$IteratorSpliterator.forEachRemaining(Unknown Source)
        at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
        at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(Unknown Source)
        at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
        at java.util.stream.ReferencePipeline.collect(Unknown Source)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.loadPluginJar(PluginLoader.java:167)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.lambda$loadAllJarsFromDir$1(PluginLoader.java:122)
        at java.util.stream.ForEachOps$ForEachOp$OfRef.accept(Unknown Source)
        at java.util.ArrayList.forEach(Unknown Source)
        at java.util.stream.SortedOps$RefSortingSink.end(Unknown Source)
        at java.util.stream.Sink$ChainedReference.end(Unknown Source)
        at java.util.stream.Sink$ChainedReference.end(Unknown Source)
        at java.util.stream.AbstractPipeline.copyInto(Unknown Source)
        at java.util.stream.AbstractPipeline.wrapAndCopyInto(Unknown Source)
        at java.util.stream.ForEachOps$ForEachOp.evaluateSequential(Unknown Source)
        at java.util.stream.ForEachOps$ForEachOp$OfRef.evaluateSequential(Unknown Source)
        at java.util.stream.AbstractPipeline.evaluate(Unknown Source)
        at java.util.stream.ReferencePipeline.forEach(Unknown Source)
        at edu.wpi.first.shuffleboard.app.plugin.PluginLoader.loadAllJarsFromDir(PluginLoader.java:120)
        at edu.wpi.first.shuffleboard.app.Shuffleboard.start(Shuffleboard.java:89)
        at com.sun.javafx.application.LauncherImpl.lambda$launchApplication1$161(LauncherImpl.java:863)
        at com.sun.javafx.application.PlatformImpl.lambda$runAndWait$174(PlatformImpl.java:326)
        at com.sun.javafx.application.PlatformImpl.lambda$null$172(PlatformImpl.java:295)
        at java.security.AccessController.doPrivileged(Native Method)
        at com.sun.javafx.application.PlatformImpl.lambda$runLater$173(PlatformImpl.java:294)
        at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
        at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
        at com.sun.glass.ui.win.WinApplication.lambda$null$147(WinApplication.java:177)
        ... 1 more
Caused by: java.lang.ClassNotFoundException: javax.jms.MessageListener
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        ... 50 more
```

I was able to go through log4j.jar with a dull machete and only include the essential .class files and get it to come up.  I'm fine doing this, but am wondering if this is just a weird edge case, or if I should so something else, or if this is a potential problem.

SamCarlberg, 2018-02-04:
Sanity check: is there a class `javax/jms/MessageListener.class` in the fatjar?

SamCarlberg, 2018-02-04:
To answer your first question, you should be able to just bundle a "fat" or "uber" jar that contains all your compiletime and runtime dependencies (exception the shuffleboard API - that's already included in the app and bringing in a different version can case Major Problems). As far as shuffleboard is concerned, those are _your_ classes. Log4J might be a weird edge-case since I believe it does some stuff with classloading. Might be confusing it with SLF4J, though.

You could also use `java.util.logging.Logger` to do all your logging. That's what we use internally, and it's [already set up to log to disk](https://github.com/wpilibsuite/shuffleboard/blob/master/app/src/main/java/edu/wpi/first/shuffleboard/app/Shuffleboard.java#L141-L173) and console.

pjreiniger, 2018-02-04:
That class is not in the fatjar.  First thing I checked.  Nothing obviously similar to it either.

I thought about using the built in logger but shied away since "the java community" seems to prefer the other libraries like log4j and slf4j.

I assumed it was an edge case based on how weird it was, but thought I should document it anyway.  Thanks for the confirmation.

closed 2018-02-04 by pjreiniger

SamCarlberg, 2018-02-04:
It looks like that class is part of Java EE, not SE. It won't be present in the shuffleboard classpath. Looks like it's an internal or transitive dependency of Log4J