wpilibsuite/shuffleboard #400: Make title bar optional

andrewda, 2018-02-05:
It would be nice to have the option to disable title bars so that they don't take up unnecessary space. If we've got a lot of information to display, every inch is valuable, so the smaller we can make some widgets the better.

AustinShalit, 2018-05-30:
Duplicate of #42

andrewda, 2018-05-30:
Should have made this more clear in my original issue, but I'm talking about the title bars on widgets here. It would be nice to be able to disable the widget titles on some specific widgets if I want, for example, a long boolean box across the top of the window. Having the option to disable the title bar of that boolean box would reduce the footprint of the widget on the screen and allow for a cleaner UI.

I _believe_ #42 is specifically referring to hiding the larger UI features and titles, not the titles of specific widgets.

PR #606 (2019-10-26): Allow widget titles to be hidden

# Overview

Add two more title display modes: 'minimal' and 'hidden'. Minimal display reduces the size of the title bars and font, while hidden will remove title bars entirely.

Widget titles are hidden on a per-tab basis. Individual widgets have no control. This follows the existing behavior of widget titles inside layouts.

Title display modes can be modified in CSS using `.widget-pane:minimal-tile-titles` and `.widget-pane:no-tile-titles`

Closes #400
Closes #586 

Alternative to #520 

# Screenshots

### Default title display (current)
[image]

### Minimal [... cut]
closed 2019-10-26 by PeterJohnson