wpilibsuite/shuffleboard #405: Shuffleboard Is unusable without an internet connection

bot190, 2018-02-10:
The new update feature doesn't work correctly in situations without internet access. If you launch shuffleboard without an internet connection it complains that it is unable to connect to the update server. The modal that this error appears in doesn't have a way of closing it. This makes shuffleboard impossible to use without an internet connection. Something that seems extremely likely when connecting to robot wifi networks.

[image]

SamCarlberg, 2018-02-10:
This is intentional behavior to prevent users from accidentally closing the dialog and wondering why there's no update information. Press the escape button to close the dialog.

bot190, 2018-02-10:
There should be a message saying that escape will close this dialog. This would require the user to read the dialog.

PR #407 (2018-02-12): Make startup update check less intrusive
The check at startup will only show a small bar at the bottom of the screen, and only when an update is available. Nothing will be displayed if no update is available, or if the server could not be connected to (eg no internet connection).

Alerts shown from a manual update check will automatically close themselves after 15 seconds. The update prompt, however, does _not_ do this and retains its previous behavior.

Closes #405 

# Screenshots

[image]
closed 2018-02-12 by SamCarlberg