wpilibsuite/shuffleboard #408: No Connection Error will not clear.

Team3840, 2018-02-12:
Trying to connect to via driver station while someone else was already connected to the robot thru Driver Station. Cause this No connection error.  However, after shutting everything done and rebooting I cannot get this error to go away.  Every time we reopen shuffle board this error reappears and even though it's connected with values updating but with the error message we not able to do anything.

[image]

SamCarlberg, 2018-02-12:
This is the update check at startup. You can disable it in the application settings, or close the dialog by pressing the Escape button on your keyboard. This was fixed with #407 and will be available in the coming days.

Also a duplicate of #405

closed 2018-02-12 by SamCarlberg

Kevin-OConnor, 2018-02-12:
That dialog is about connection to the internet for auto update, not connection to the robot. Press escape to dismiss the dialog box. We are working on a shuffleboard update which will make the update checker less intrusive