wpilibsuite/shuffleboard #413: Problem with Opening Save FIles

PCApple, 2018-02-16:
We have been using the shuffleboard for about a week now. Recently, we have been unable to open saved versions of shuffleboard layouts. We are using shuffleboard 1.2. It also will not open any tabs and when it does, it's completely blank.

SamCarlberg, 2018-02-16:
Please share the save files you're trying to load

PCApple, 2018-02-16:
[save file.zip](https://github.com/wpilibsuite/shuffleboard/files/1729910/save.file.zip)

SamCarlberg, 2018-02-16:
Your save file looks okay; I can load it on my machine. These problems are usually due to incompatible or invalid save files.

Are any exceptions or warnings present in the log file?

sataylor000, 2018-02-17:
I'm seeing a similar issue with several of our team laptops-- they open Shuffleboard, but no tabs are present.  I can drag widgets onto a new tab, and save the file.  When I exit and restart, the saved setup is not restored by any means (even though I manually saved, AND had show confirmation dialog on exit (and let it save), AND had load on startup set, AND tried loading the save file manually.)

This seems to be persistent, but I need to recheck tomorrow.  We are at WPI all weekend finishing the robot, so if someone happens to be around and wants to see a live repro, drop by The Pits.  I'll do some more experiments and see if I can characterize it better tomorrow when I have access to the laptops again.

PCApple, 2018-02-17:
Hello,
I sorry for not responding in a while, I attached the log file
[shuffleboard.log](https://github.com/wpilibsuite/shuffleboard/files/1733923/shuffleboard.log)

PCApple, 2018-02-17:
Update:
I was able to fix the Warning could not load Prefs err, but ran into an error with the loading, here is the log file
[shuffleboard.log](https://github.com/wpilibsuite/shuffleboard/files/1733959/shuffleboard.log)

SamCarlberg, 2018-03-03:
@PCApple It looks like you're running into #358, which was fixed in #384. Please make sure you're using at least version 1.2.0

@sataylor000 Please also make sure you're using version 1.2.0 or higher

If this behavior still exists in 1.2.0 or 1.2.1, please upload the log files

sataylor000, 2018-03-04:
I checked my logfiles yesterday afternoon, and sure enough there was an error.  However, the robot was unbagged and I didn't have a chance to capture the exact text.  That being said, the error was pretty clear:  it said that the save file was using a tile size of 8, but that the minimum tile size allowed was 32.  Sure enough, the default in Shuffleboard under both Preferences and Current Tab Preferences was set to 8, so every save file we created (and shared between systems) was slowly corrupting all the systems.  We had set the tile spacing as small as the tool would allow because we're trying to squeeze content onto a Classmate as our DriverStation, and we can't even see two widgets vertically above the DS window. 
Blowing everything away and starting over allowed us to get things working again (but now I don't have the logfile)

This begs the question of how part of the tool believed 8 was legal, while another part did not.  Should I duplicate this comment as a separate issue?