wpilibsuite/shuffleboard #414: Extremely resource intensive and occasionally freezing

aswerdlow935, 2018-02-16:
Running on a relatively new Driver Station Laptop (u series 6th gen dual core), Shuffleboard has become extremely resource intensive, often freezing. Used to be fine, but is now extremely laggy after the new update with frequent complete freezes and slowdowns. Running with a total of ~30 network table values (without about 8 as graphs). Smartdashboard doesn't seem to have this problem, and it does not appear to be caused by other system processes.

SamCarlberg, 2018-02-16:
Remove or reduce the size of the graphs; they are very resource intensive, especially when they're large or have a lot of data points

19lmyers, 2018-02-16:
This is happening to me as well, and did not happen on the previous
version. Is there a way to fix it without removing the graphs?

bradamiller, 2018-02-16:
The previous version of Shuffleboard worked and it got worse with the latest update?

19lmyers, 2018-02-16:
That's exactly what happened to me.

aswerdlow935, 2018-02-20:
Note that version 1.0.0 does not seem to have this problem

JohnGilb, 2018-02-28:
I am also seeing this issue - no graph used, just 6 tabs with about 8 properties each.

allengregoryiv, 2018-03-02:
Version 1.2.1 I'm getting extremely high memory usage (1 gig to 1.7 gigs) after a few minutes of being connected to the robot. Once this happens I have to close the java process in the task manager. It may be the graphs doing this but that's still a bit of a pain when we need the graphs for the tuning.

allengregoryiv, 2018-03-07:
Any word on a fix for this issue? Trying to figure out if we are going to be able to use shuffleboard during competition this year.

andrewda, 2018-03-07:
Also having extremely high memory usage here. Terminating the Java process seems to work for a few minutes, but then causes the computer to run incredibly slowly. This is a pretty serious issue for us and a limiting factor for our use of Shuffleboard at competitions.

SamCarlberg, 2018-03-07:
@allengregoryiv and @andrewda please share your save files and a network tables snapshot (can be taken with OutlineViewer). If you've set a custom network tables update rate, share that as well.

andrewda, 2018-03-07:
@SamCarlberg Here's the Shuffleboard save file: https://gist.github.com/andrewda/7b4f8b532ff160838fa03fe64534c2e6. Will send you a snapshot when I have access to the robot in about an hour. Using default network tables update rate.

SamCarlberg, 2018-03-08:
@allengregoryiv how many values are you sending over networktables? I can only see recording using that much memory so fast if there are hundreds of values being sent, or if you've cranked the update rate way up

alexanderswerdlow, 2018-03-08:
We're running about 20 (without any extra plugins), and directly after cloning and running we get the same behavior (with recording both on/off). Maybe would erasing all shuffleboard data fix things? (As in not just in the repo dir but where shuffleboard saves)

allengregoryiv, 2018-03-08:
The extreme cases were during PID tuning, running a graph for position, error and output. Even just sitting with the same instance open where we were just testing basic robot pneumatic functions it's at 800 mbs.

allengregoryiv, 2018-03-08:
Shuffleboard save file - https://drive.google.com/open?id=1Lcse-tgw9lYPYGtiI4I6yrK6H9FapdkZ

Shuffleboard recording - https://drive.google.com/open?id=1dqs1Mdk9QIaY9hV3ZhonKTq5EChAA3_N

Not near the robot to get a snapshot, I'll try tomorrow.

Shuffleboard is taking 880mb and running. This instance has been up for a couple hours though.

SamCarlberg, 2018-03-08:
Long-running recording instances can take a lot of memory. That's something I'll have to fix in the offseason. For now, you can disable recording, or simply start a new one and let the garbage collector free up the memory from the old recording.

allengregoryiv, 2018-03-08:
Is there a way to not default to recording on?

SamCarlberg, 2018-03-08:
No

allengregoryiv, 2018-03-08:
It seems like just leaving a session open, recording or not keeps leaking memory. Recording has been off for a while and I'm at 900mb on the that session now.

falsephoenix, 2018-03-09:
I just want to say that this issue caused excessive packet loss when connected to FMS during a competition, to the point of a jittering drivetrain. 

It would be helpful to at least provide some documentation discouraging the use of graphs until further notice, in case other teams make the same mistake I did.

SamCarlberg, 2018-03-16:
Fixes have been merged to improve the performance of graphs and long recordings and will be available in the next release

closed 2018-03-16 by SamCarlberg