wpilibsuite/shuffleboard #416: File menu contents render but are not always persistent or selectable

sataylor000, 2018-02-17:
I have a system that pretty frequently gets into a state where the File menu can be clicked and unrolls, but the menu disappears as soon as I release the mouse button, and I am unable to click on any File menu entries.  In some cases, the menu also does not roll down from the menu bar, but shows up about halfway down the window, like this:
[image]

I will be at WPI all weekend in the Pits, in case someone wants to see a live repro of this issue.  I am only seeing it on one specific laptop thus far, and not every time I start shuffleboard.  I see it happen more often at home when I have an external monitor attached, but the "unable to click" happens on the bare laptop as well.  This is using v1.2.0, but I saw it on a previous version as well (1.0?)

sataylor000, 2018-03-03:
I discovered at least part of the reason for the offset menu.  The monitor layout I use for this particular system has a second monitor "virtually above" the first.  When that monitor is connected, the apparent virtual vertical size of the desktop is taller and the menu shifts down.  If I disconnect that monitor, the menu item ends up in the right place.  Could the code be scaling something for the position of the menu based on desktop height and not window height...?

SamCarlberg, 2018-03-03:
If that's the case, this is a bug in JavaFX. I'll look into workarounds