wpilibsuite/shuffleboard #417: ComboBox Chooser Selection does not visually update

benpesposito, 2018-02-17:
For the ComboBox Choose, the selected value will change, which can be seen if I print out the .getSelected(), but it will not visually update on shuffleboard.

sciencewhiz, 2018-02-18:
How is the selected value changing?

scottdavis2052, 2018-02-19:
We have seen this multiple times.  We have our auto modes in a drop down chooser. The driver will select a different auto mode from the list, but the text in the top (the selected item) does not change. If you stop and restart shuffleboard, the value you had previously selected, now shows as the selected item.  

The shuffleboard interface does seem to be changing the network table value correctly, but it does not always update the UI.  As a result, a team will think they are going to run one automode based on the interface, but run another based on the network table value.

These seems to happen after multiple times the robot is powered off/on, while shuffleboard remains open. However, we have not found a way to consistently reproduce.

SamCarlberg, 2018-02-20:
Due to how NetworkTables does synchronization, selecting a value from a chooser (or otherwise send data to the robot), then restarting the robot program, and then later restarting shuffleboard, the value previously sent from the dashboard will disappear from NetworkTables.

The best workaround for now would be to always select the autonomous mode before every match. You can verify that it's been sent by looking at the NetworkTable overview in the sources pane.

andrewda, 2018-02-21:
@SamCarlberg from what I've seen, this is more of a visual bug in Shuffleboard as opposed to a synchronization issue. After clicking on a new value in the dropdown menu, the value displayed stays unchanged, but if the dropdown is opened again, the newly set value is highlighted correctly.