wpilibsuite/shuffleboard #418: ArrayIndexOutOfBoundsException: -1 in JavaFX

auscompgeek, 2018-02-18:
… not really sure what's happening here. The traceback doesn't seem very useful either.

```
edu.wpi.first.shuffleboard.app.Shuffleboard uncaughtException
WARNING: Uncaught exception on JavaFX Application Thread
java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:418)
	at java.util.ArrayList.get(ArrayList.java:431)
	at com.sun.javafx.collections.ObservableListWrapper.get(ObservableListWrapper.java:89)
	at com.sun.javafx.collections.VetoableListDecorator.get(VetoableListDecorator.java:306)
	at javafx.scene.Parent.updateCachedBounds(Parent.java:1591)
	at javafx.scene.Parent.recomputeBounds(Parent.java:1535)
	at javafx.scene.Parent.impl_computeGeomBounds(Parent.java:1388)
	at javafx.scene.layout.Region.impl_computeGeomBounds(Region.java:3078)
	at javafx.scene.Node.updateGeomBounds(Node.java:3579)
	at javafx.scene.Node.getGeomBounds(Node.java:3532)
	at javafx.scene.Node.getLocalBounds(Node.java:3480)
	at javafx.scene.Node.updateTxBounds(Node.java:3643)
	at javafx.scene.Node.getTransformedBounds(Node.java:3426)
	at javafx.scene.Node.updateBounds(Node.java:559)
	at javafx.scene.Parent.updateBounds(Parent.java:1719)
	at javafx.scene.Parent.updateBounds(Parent.java:1717)
	at javafx.scene.Scene$ScenePulseListener.pulse(Scene.java:2404)
	at com.sun.javafx.tk.Toolkit.lambda$runPulse$3(Toolkit.java:355)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.sun.javafx.tk.Toolkit.runPulse(Toolkit.java:354)
	at com.sun.javafx.tk.Toolkit.firePulse(Toolkit.java:381)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:510)
	at com.sun.javafx.tk.quantum.QuantumToolkit.pulse(QuantumToolkit.java:490)
	at com.sun.javafx.tk.quantum.QuantumToolkit.lambda$runToolkit$11(QuantumToolkit.java:319)
	at com.sun.glass.ui.InvokeLaterDispatcher$Future.run(InvokeLaterDispatcher.java:95)
	at com.sun.glass.ui.gtk.GtkApplication._runLoop(Native Method)
	at com.sun.glass.ui.gtk.GtkApplication.lambda$null$5(GtkApplication.java:139)
	at java.lang.Thread.run(Thread.java:748)
```

auscompgeek, 2018-02-18:
Other symptoms: the UI appears to freeze, however it still responds to inputs; it only stops rendering.

SamCarlberg, 2018-02-22:
There is probably an exception thrown earlier by shuffleboard. Can you share the full log file?

AustinShalit, 2018-05-30:
Closing for now.  @auscompgeek If you have the full log file, please post it and reopen.

closed 2018-05-30 by AustinShalit