wpilibsuite/shuffleboard #421: Shuffleboard Crash when Velocity Mode is Used

GaganBhat, 2018-02-18:
Shuffleboard works totally fine for everything else....except when I do velocity mode on the robot. I am using CTRE v4 Legacy for my talons. 

The moment the robot drives in velocity mode, shuffleboard lags and becomes unresponsive. It does give me the dialogue box asking me if it wants to save the current setup when I try to close it though. 

We are not sending any velocity related values to shuffleboard.

sciencewhiz, 2018-02-18:
Do you have a minimal robot code test case that demonstrates this?

GaganBhat, 2018-02-20:
Yup, the moment I change the talon mode to velocity, for some odd reason, it freezes lags and eats up the CPU of the driverstation through....shuffleboard.

SamCarlberg, 2018-02-20:
Does closing shuffleboard significantly decrease CPU usage?

GaganBhat, 2018-02-20:
Nope, in the background a java process continues to make the entire driverstation lag and makes the robot uncontrollable. This has been tested on multiple laptops.

CPU usage is around 90% by that one process.

SamCarlberg, 2018-02-20:
Can you confirm that the Java process is the shuffleboard process? Check the process ID when you start shuffleboard, then confirm that it's the same as the one that pegs your CPU.

If it _is_ the shuffleboard process, we'd need you to open OutlineViewer and save the NetworkTable state and share that .ini file so we can see what's going on in NetworkTables. It would also be useful to have the shuffleboard log file as well

bigabos-hey, 2018-02-22:
Sam,

Did you get the information you needed to help resolve this?  I believe we are experiencing the same bug.

SamCarlberg, 2018-04-02:
I am not familiar with CTRE's NetworkTables telemetry implementation. It's odd that this only happens when the Talons are in velocity mode. You could take a snapshot of the NetworkTables state with OutlineViewer, though I suspect the cause is elsewhere.

For sanity's sake, does this still happen with version 1.3.1?

GaganBhat, 2018-04-05:
I will check tonight and let you know. Installing 1.3.1 right now...

GaganBhat, 2018-04-05:
Just ran that one command again. It did the exact same thing. Completely froze. The only thing different about this time though is that it recovered after around 20 seconds.

bradamiller, 2018-04-06:
Does it also do that with SmartDashboard?

GaganBhat, 2018-04-06:
Nope, it doesn't happen with SmartDasboard.

SamCarlberg, 2018-04-09:
@GaganBhat Please take a "before" and "after" snapshot of the NetworkTables state with OutlineViewer. You should also have a data recording from shuffleboard with everything up to the point of the crash; please also share that as well. Having a copy of your robot code will also make it a lot easier to troubleshoot. You can upload each of those to this issue thread in a .zip, or send them to me in an email if you don't want them public

GaganBhat, 2018-04-09:
Absolutely. I will do that today and attach it to this thread. By "snapshot" you mean a simple screenshot of contents of the network tables displayed in the left side of the program right?

SamCarlberg, 2018-04-09:
No, an actual content dump. You can do that with File->Save State. That will create a .ini file containing everything in NetworkTables at the time you save it

bradamiller, 2018-04-10:
Can you zip the code for a minimal failing example and post it here? One of the CTRE developers offered to look at it to see what's up. We can then also test it with Shuffleboard.

GaganBhat, 2018-04-22:
Sorry for the delayed response, 

For the past week I have been trying to get the post-crash data, but Shuffleboard just doesn't respond anymore. I also couldn't find a "save state" option so I just saved it like I normally do. Here's the ZIP file with the command I am running, as well as the latest log files where I experienced the crash. 

@bradamiller Although it's already in the zip file attached, here's a quick link to the minimal failing example. [https://gist.github.com/GaganBhat/e64476c7b9ed5c256613572febfad156](url)

[logs_code.zip](https://github.com/wpilibsuite/shuffleboard/files/1935709/logs_code.zip)

sciencewhiz, 2018-04-22:
Sam is asking you to use the OutlineViewer tool that also comes with the wpilib plugins. Open both Shuffleboard and OutlineViewer and when Shuffleboard crashes, use OutlinrViewer to save the state.

GaganBhat, 2018-04-22:
[pre_post_save_states.zip](https://github.com/wpilibsuite/shuffleboard/files/1935981/pre_post_save_states.zip)

@sciencewhiz Thanks! I have attached the state of the network tables using the Outline Viewer in the zip file above.
It has the states right before the crash as well as right after the crash.

It also includes the code for the command that is causing the crash.