wpilibsuite/shuffleboard #426: Feature Request: Option to disable recording by default

allengregoryiv, 2018-03-08:
Can we get an option in preferences to disable recording by default. Recording seems to take a ton of ram when left running and we rarely want to be recording most things. 

Or can recording be turned on and off from a networktables key? Then we could start recording during FMS connected matches.

iammarix, 2018-03-13:
We were running into the same issue, and thought a more elegant solution was a plugin (rather than our initial "find the line in shuffleboard source and comment it out, then build our own jar" idea).

https://github.com/teamhyper/shuffleboardNoRecord

SamCarlberg, 2018-03-13:
The problem with disabling recording by default is that you have to remember to turn it on every time you want to use it; it's very easy to forget and be hosed if you need to look back at the data.

#432 fixes the root problem by significantly reducing the memory and CPU overhead of the recording process. You'll be seeing much less memory and GC pressure when that PR gets merged.

AustinShalit, 2018-05-30:
Agree.  There is no way to disable LV logging and we should follow them.

closed 2018-05-30 by AustinShalit