wpilibsuite/shuffleboard #428: Graph widgets take a lot of CPU and memory

SamCarlberg, 2018-03-11:
High CPU use is from the render thread re-rendering the entire area chart series on each pass, combined with a high update rate from data sources. Decreasing the update rate and doing batch updates (updating with many values at once) and caching the drawn series shape can significantly reduce render thread usage.

High memory use is from old values being stored as `XYChart.Data`, each of which takes up 216 bytes, plus 128 bytes for the data itself. As time progresses, this old data can balloon to take up hundreds of megabytes. Storing old data in primitive arrays (eg `double[] xValues` and `double[] yValues`) can cut this down to an amortized 128 bytes per data point.

bradamiller, 2018-03-11:
Does the XYChart.Data get reallocated often? Is that the problem? Maybe creating a pool and managing it yourself would save a lot of time.

SamCarlberg, 2018-03-11:
The problem is that, after a few hours, hundreds of thousands of these data objects can be kept around without being visible. This uses a lot of memory and causes GC pressure.

For example, an `ArrayList<XYChart.Data<Number, Number>>` takes:

- _n_ * 8 bytes for each data point in the list (for references in the array) (amortized)
- At least ~330* bytes per data point

Totals to at least _n_ * 338 bytes. For a value that updates at 10Hz for 1 hour, this comes to be at least **12,168,000** bytes

Versus parallel primitive arrays:

- 8 bytes per X value (amortized)
- 8 bytes per Y value (amortized)

Totals to _n_ * 16 bytes. For a value that updates at 10Hz for one hour, this comes out to be **576,000** bytes, or ~21x less than a list of `XYChart.Data`

Multiply these numbers by the number of graphed values and the savings become enormous. 10 graphs becomes 121,680,000 bytes, vs 5,760,000 bytes for the primitive arrays. This is a _huge_ improvement, especially for systems with less than 2GB of RAM.

<sup>* I probably missed a few object references in the tree for `XYChart.Data`, so that number could easily be larger than 330 bytes.</sup>

PR #430 (2018-03-16): Improve CPU and memory usage of graphs
Fixes #428 

Stores the "real" data in parallel primitive `double[]` arrays to reduce memory footprint by a factor of 20.
Does 4Hz batch updates to graph data to reduce the amount of render calls
Caches rendered series nodes to speed up rendering

Overall reduces CPU use by ~30-40% for rendering graphs on both my desktop (i7-7700K @ 4.2GHz) and laptop (i7-6700HQ @ 2.6GHz) - render thread use drops from 95-100% use down to 60-70%
closed 2018-03-16 by SamCarlberg