wpilibsuite/shuffleboard #443: Camera's don't auto-reconnect.

Daltz333, 2018-03-18:
If the DriverStation goes to sleep with cameras connected, and then disconnected. Shuffleboard will not re-get the camera stream on re-connection. It's fixed by closing Shuffleboard and reopening it, but it's still annoying.

Daltz333, 2018-03-18:
Some more information on my setup. Robot network is all static IPed, and camera servers on running on a Raspberry Pi3 using PyCScore. It works flawless in SmartDashboard and via the browser, and it only pops up occasionally using Shuffleboard.

Daltz333, 2018-04-05:
Can I get a confirmation that I'm not the only one who sees this? I have replicated this on my personal laptop as well as two driver station laptops.

Daltz333, 2018-09-12:
Any updates @SamCarlberg on this issue?

SamCarlberg, 2018-09-13:
It's getting fixed upstream in cscore

closed 2018-12-29 by PeterJohnson