wpilibsuite/shuffleboard #444: Slow startup on some machines

SamCarlberg, 2018-03-19:
Some users report very slow startup times, up to several minutes. This is Not Good.

For reference, it takes 4-5 seconds to load on my laptop (i7-6700HQ, mid-range SSD) and ~2.8 seconds on my desktop (i7-7700k, 960 Pro NVMe SSD).

In the past, I've seen the camera server plugin take a long time to load. I have not seen that behavior for a while, though

JLLeitschuh, 2018-03-19:
Asynchronous loading of some things perhaps?

SamCarlberg, 2018-03-19:
Do you mean that some things are being loaded asynchronously that shouldn't be, or that some things should be async that currently aren't?

JLLeitschuh, 2018-03-19:
If everything is loaded in the thread that calls `start` that might not be the best idea.

AustinShalit, 2018-05-30:
Closing as #466 has made this much better.

closed 2018-05-30 by AustinShalit