wpilibsuite/shuffleboard #453: Unable to add robot preferences

nix342, 2018-03-24:
We could not figure out how to add robot preferences in shuffleboard. We did see a robot preferences e source and could view that, but saw no way to add new preferences as you can in smartdashboard. See the doc here for a description of the feature in smartdashboard:
https://wpilib.screenstepslive.com/s/currentCS/m/smartdashboard/l/255423-setting-robot-preferences-from-smartdashboard

Is this feature supported in shuffleboard?

SamCarlberg, 2018-03-26:
Robot preferences won't show up in the default tabs because it's stored in the `/Preferences` table and hte default tabs only autopopulate with the `/Smartdashboard` and `/Livewindow` tables.

You should see a subtree in the NetworkTables source tree for the preferences; you can drag this into a tab to create a preferences widget.

nix342, 2018-03-26:
We were able to do that, but once we did, we couldn't figure out any way to add a new preference...the widget was blank with the exception of a search field. We entered values into the search field, but that didn't do anything.

SamCarlberg, 2018-03-26:
You need to add preferences in your robot program. It doesn't make sense to be able to define preferences dashboard-side because the program will have to know about them beforehand

nix342, 2018-03-26:
OK, we can do that, but that wasn't necessary in the past when using smartdashboard. The robot preference widget there allows new preferences to be added dashboard-side. We have only ever used the getter functions for preferences in our robot code.

nix342, 2018-04-09:
After thinking about this more, I do believe the ability to add and edit preferences on the dashboard is an important feature. We make extensive use of these, particularly in our autonomous coding where we are able to define steps and check a preference value entered in the dashboard to see how many steps we should execute. We also use them for defining distances and headings in order to fine tune complex paths. Outside of autonomous, we use them to quickly update limits for subsystems, making it easy to "zero" a mechanism quickly without changing code. Generally, any area of the code which we want to iterate on quickly can be handled very well using a preference.

While all of these could be set up and predefined in robot code, it is cumbersome as you need to check to see if the preference is present before setting it up, otherwise you will lose any previously saved data. It is just much easier to enter them as you need them via the dashboard. 

So, I would ask you to reconsider and allow preferences to be added and edited on the dashboard side.

Thanks!

AustinShalit, 2018-05-30:
@bradamiller Do you have an opinion on this?

bradamiller, 2018-05-30:
I do. I agree that it would be nice to create preferences on the dashboard side even if they haven't been set yet on the robot. Imagine you are adding some constants. You add then to NT, then add the code to read them on the robot. Especially if the workflow is to only ever set them with the dashboard and always read with the robot - something that doesn't seem to be a unreasonable use of preferences.

SamCarlberg, 2018-08-23:
@nix342 I think this has been solved with #492 and wpilibsuite/allwpilib#1022. 

You will be able to define properties in your robot programs that will be placed in Shuffleboard automatically; for a basic example:
```java
private final NetworkTableEntry armPosition;

@Override
public void robotInit() {
  armPosition = Shuffleboard.getTab("Drivers").add("Arm Position", 1.0);
  // If you want the value to be saved when the robot shuts down:
  armPosition.setPersistent();
}

@Override
public void autonomousInit() {
  double armPos = armPosition.getDouble(1.0);
  // Do something with the arm position
}
```
It's an alternative API to RobotPreferences; however, you can get identical behavior by calling `setPersistent()` on the NetworkTable entries.

It also gives you greater flexibility in how your preferences are organized and grouped in Shuffleboard; for example, grouping preferences in layouts on a per-subsystem basis.

nix342, 2018-08-28:
This is ok, but it is basically just the same answer as given before. A bunch of code is still needed in robotInit that would not otherwise be needed if preferences could be created on the dashboard side, as they can with SmartDashboard.

In addition, I think there may be an issue with your sample code. Even though the item is marked as persistent, it gets overwritten every time robotInit is called and the item is added. I think you need to add a check to see if the item exists, like this:

```
@Override
public void robotInit() {
  if (armPosition.getDouble(1.0) == 1.0) {
    // if value is 1.0 it is either set to 1.0 or doesn't exist and we're getting the default.
    // either way, it is safe to (re)initialize
    armPosition = Shuffleboard.getTab("Drivers").add("Arm Position", 1.0);
    // If you want the value to be saved when the robot shuts down:
    armPosition.setPersistent();
  }
}
```
Do I understand how that works correctly? Is that check necessary to avoid overwriting any persistent value?

SamCarlberg, 2018-08-28:
You don't need to check if a value is already present. The value in `add` is the default value to use if it hasn't been set, similar to how you have to specify a default value when getting a value with `SmartDashboard.getX(key, defaultValue);`.

So `Shuffleboard.getTab("Drivers").add("Arm Position", 1.0)` will _only_ set the value of the entry if the entry does not already exist. Existing values are not overwritten.

A convenience method `addPersistent` has also been added that will make the value of the entry persistent:
```java
@Override
public void robotInit() {
  armPosition = Shuffleboard.getTab("Drivers").addPersistent("Arm Position", 1.0);
}
```

The pull request to WPILib has not been merged yet, which is why you can't find any API documentation for it.

nix342, 2018-08-29:
OK, not having to check if it exists is good. This essentially adds one line, which is not too painful.

I do still have some concerns, not the least of which is that we are tied to Shuffleboard with this method. We'll experiment with using Shuffleboard in this way in the pre-season, but it will be difficult to commit to this method for the competition season unless we notice a marked improvement in overall stability.

#492 is a very cool feature, btw. It appears that it will vastly improve building of Shuffleboard tabs...we experimented last year with creating a tab per subsystem, and that was good...but with this feature we can get better and more consistent organization.

I would still like to see the ability to add robot preferences on the dashboard side, though. If that's not on the roadmap for Shuffleboard, however, I can accept that.

carbotaniuman, 2020-09-06:
Looks OBE. Disposition to close @PeterJohnson

closed 2020-09-10 by AustinShalit